cyberbotics/webots_ros2 #18: Missing controller module for webots_ros2_universal_robot

ruffsl, 2019-09-06:
I was just testing out the release of the `webots_ros2_universal_robot package` but encountered a `ModuleNotFoundError`, so perhaps there's a missing in depencies declarations or release packaging? 

https://discourse.ros.org/t/new-packages-for-ros-2-crystal-2019-09-05/10562

* OS: Ubuntu 18.04
* ROS distro: crystal
* type: release install

Installed via:

```
apt install ros-crystal-webots-ros2*
```

Runtime error:

<details>

```
$ ros2 launch webots_ros2_universal_robot universal_robot.launch.py 
[INFO] [launch]: process[webots_launcher-1]: started with pid [10099]
[INFO] [launch]: process[universal_robot-2]: started with pid [10100]
Traceback (most recent call last):
  File "/opt/ros/crystal/lib/webots_ros2_core/webots_launcher", line 11, in <module>
    load_entry_point('webots-ros2-core==0.0.2', 'console_scripts', 'webots_launcher')()
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_core/webots_launcher.py", line 42, in main
    subprocess.call(command)
  File "/usr/lib/python3.6/subprocess.py", line 287, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/opt/ros/crystal/share/webots_ros2_desktop/webots/webots': '/opt/ros/crystal/share/webots_ros2_desktop/webots/webots'
[ERROR] [launch]: process[webots_launcher-1] process has died [pid 10099, exit code 1, cmd '/opt/ros/crystal/lib/webots_ros2_core/webots_launcher --mode=realtime --world=/opt/ros/crystal/share/webots_ros2_universal_robot/worlds/universal_robot.wbt'].
[INFO] [launch]: sending signal 'SIGINT' to process[universal_robot-2]
"WEBOTS_HOME" is not correctly set.Traceback (most recent call last):
  File "/opt/ros/crystal/lib/webots_ros2_universal_robot/universal_robot", line 11, in <module>
    load_entry_point('webots-ros2-universal-robot==0.0.2', 'console_scripts', 'universal_robot')()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 487, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2728, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2346, in load
    return self.resolve()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 17, in <module>
    from webots_ros2_core.webots_node import WebotsNode
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_core/webots_node.py", line 34, in <module>
    raise e
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_core/webots_node.py", line 31, in <module>
    from controller import Robot
ModuleNotFoundError: No module named 'controller'
[ERROR] [launch]: process[universal_robot-2] process has died [pid 10100, exit code 1, cmd '/opt/ros/crystal/lib/webots_ros2_universal_robot/universal_robot'].
```

DavidMansolino, 2019-09-06:
Hi, nice to see that people are already testing this package.
In addition to the ROS packages, you probably need to install Webots as explained in the documentation: http://wiki.ros.org/webots_ros2#Webots_Software, then you have to define the ``ROS2_WEBOTS_HOME `` environment variable to point to the Webots installation directory.
We are trying to improve this and embed the Webots software in the package so that you don't have to install it for a future version of the packages.

ruffsl, 2019-09-06:
> We are trying to improve this and embed the Webots software in the package so that you don't have to install it for a future version of the packages.

Ok, yea that `webots_2019b_amd64.deb` at 1.3 GB is rather large. When you do get to packaging webots, might want to break out the bigger assets into separate packages so folks only need to install what they declare they depend on.

So I've installed webots from the release deb above, but this line in the wiki doesn't seem to hold:

> 4. If none of the previous point is set/installed ROS2 will look for Webots in the default installation paths (e.g. /usr/local/webots).

http://wiki.ros.org/webots_ros2#Webots_Software

So as you suggested, I've manually set the environment variable:

```
export ROS2_WEBOTS_HOME=/usr/local/webots
```

Now webots window pops up during launch, but it looks like the `control_msgs` is still missing the action message type in crystal:

```
$ ls /opt/ros/crystal/share/control_msgs
cmake  environment  local_setup.bash  local_setup.sh  local_setup.zsh  msg  package.xml  srv
```

<details>

```
$ ros2 launch webots_ros2_universal_robot universal_robot.launch.py 
[INFO] [launch]: process[webots_launcher-1]: started with pid [17718]
[INFO] [launch]: process[universal_robot-2]: started with pid [17719]
Traceback (most recent call last):
  File "/opt/ros/crystal/lib/webots_ros2_universal_robot/universal_robot", line 11, in <module>
    load_entry_point('webots-ros2-universal-robot==0.0.2', 'console_scripts', 'universal_robot')()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 487, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2728, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2346, in load
    return self.resolve()
  File "/usr/local/lib/python3.6/dist-packages/pkg_resources/__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 19, in <module>
    from webots_ros2_core.trajectory_follower import TrajectoryFollower
  File "/opt/ros/crystal/lib/python3.6/site-packages/webots_ros2_core/trajectory_follower.py", line 23, in <module>
    from control_msgs.action import FollowJointTrajectory
ModuleNotFoundError: No module named 'control_msgs.action'
[ERROR] [launch]: process[universal_robot-2] process has died [pid 17719, exit code 1, cmd '/opt/ros/crystal/lib/webots_ros2_universal_robot/universal_robot'].
```

</details>

Looks like its in the repo, but perhaps never got synced?

https://github.com/ros-controls/control_msgs/blob/2f176334e2b4d8fa88f3a54681bc033df58037b4/control_msgs/CMakeLists.txt#L34-L40

I see you've also blocked on `control_msgs` for releasing into dashing, so perhaps ping the maintainer? cc @bmagyar

DavidMansolino, 2019-09-06:
> Ok, yea that webots_2019b_amd64.deb at 1.3 GB is rather large. When you do get to packaging webots, might want to break out the bigger assets into separate packages so folks only need to install what they declare they depend on.

Yes indeed that's the current problem we are facing. and is still work in progress.

> So I've installed webots from the release deb above, but this line in the wiki doesn't seem to hold
> If none of the previous point is set/installed ROS2 will look for Webots in the default installation paths (e.g. /usr/local/webots).

Unfortunately, this is recent development which is not included in the just released package (it took a few weeks to sync the package and we did add this mechanism in the meantime, but this will for sure work in the next version of the packages).

> Now webots window pops up during launch, but it looks like the control_msgs is still missing the action message type in crystal:

You are completely right, they did a recent fix about the missing actions but never sync it:
   https://github.com/ros-controls/control_msgs/pull/27

I just asked them for a sync this morning:
  https://github.com/ros-gbp/control_msgs-release/issues/3

If you want you can get the ros-control package and compile it from sources on this branch (this is what I am doing for testing for dashing): 
  https://github.com/ros-controls/control_msgs/tree/crystal-devel

ruffsl, 2019-09-14:
I built the 12d55239c3f7010e73470615d70a26b6b145706a from source against the latest dashing sync, and got the single arm demo working, but the multi UR demo seems to fail to connect after webots succsufuly loads the world and "Finalizing nodes":

```
$ ros2 launch webots_ros2_universal_robot universal_robot_multiple.launch.py
[INFO] [launch]: All log files can be found below /home/ruffsl/.ros/log/2019-09-13-18-32-59-608674-cog-rw07-18657
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots_launcher-1]: process started with pid [18685]
[INFO] [universal_robot-2]: process started with pid [18686]
[INFO] [universal_robot-3]: process started with pid [18687]
[universal_robot-2] Cannot open file: /tmp/webots-18713/WEBOTS_SERVER
[universal_robot-2] Cannot connect to Webots: no pipe defined
[universal_robot-3] Cannot open file: /tmp/webots-18713/WEBOTS_SERVER
[universal_robot-3] Cannot connect to Webots: no pipe defined
[ERROR] [universal_robot-2]: process has died [pid 18686, exit code -11, cmd '/home/ruffsl/ws/cyberbotics/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot --webots-robot-name=UR3e __ns:=/UR3e'].
[ERROR] [universal_robot-3]: process has died [pid 18687, exit code -11, cmd '/home/ruffsl/ws/cyberbotics/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot --webots-robot-name=UR5e __ns:=/UR5e'].
```

DavidMansolino, 2019-09-14:
Thank you again for the test and good news for the single arm robot.
About the multi UR demo, it seems your are experiencing #3, using the currently released version of Webots there is one problematic constraint that is that Webots need to be started before the node for the communication between Webots and the nodes to work (which is obviously not the case since Webots takes longer to load than a simple ROS2 node).
Therefore we did add a dummy wait in the nodes:
https://github.com/cyberbotics/webots_ros2/blob/8d21a2a94fc9640d429b67ce86838c5c8bef07ea/webots_ros2/universal_robot.py#L47
However, on some computer this wait is not sufficient enough.
The fix was already implemented on the Webots side (cyberbotics/webots#793) but will be available only in the next release of Webots.
In the meantime, if you want you can try with a nightly build containing this patch (or increase the wait in the source code).
You can find nightly builds here (version R2019b-rev1): https://github.com/cyberbotics/webots/releases

Your tests and feedbacks are really appreciated!

DavidMansolino, 2019-11-07:
Fixed in Webots R2019b-rev1

closed 2019-11-07 by DavidMansolino