cyberbotics/webots_ros2 #34: Interact with the Robot UR- Never returns Goal Succeeded

LuisYordano, 2019-10-04:
Hi,

I am testing Webots with ROS2, the tutorial universal_robot:

[image]

When execute the trajectory :

`ros2 action send_goal /follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{
  trajectory: {
    joint_names: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint],
    points: [
      { positions: [-1.01, 0.38, -0.63, -0.88, 0.25, -1.63], velocities: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0], accelerations: [1.0, 1.0, 1.0, 1.0, 1.0, 1.0], time_from_start: { sec: 10, nanosec: 500 } }     
    ]
  }
}"`

The  file **trajectory_follower.py**, never execute this part of the code: :

[image]

The state is always Goal Accepted and does not change to Goal Succeeded.

Thanks in advance for your answer.

DavidMansolino, 2019-10-07:
@LuisYordano, thank you for this precise bug report, I can indeed reproduce this.
We will try to understand what is happening and fix it for the next update of the package.
I will let you know here when we found something so that if you want you can test the fix.

LuisYordano, 2019-10-07:
Ok @DavidMansolino , I wait for the fix to testing the package again.

DavidMansolino, 2019-10-08:
This should now be fixed in #36, @LuisYordano can you confirm the problem is now fixed?

LuisYordano, 2019-10-08:
@DavidMansolino , Now the trajectory ends successfully.

PR #36 (2019-10-08): Fixed Universal Robot Action Manager
Fix  #34
closed 2019-10-08 by LuisYordano