cyberbotics/webots_ros2 #38: Reset GUI Webots : webots_ros2 trayectory not responding

LuisYordano, 2019-10-10:
Hi,

I am testing Webots with ROS2, the tutorial universal_robot:

**Robot UR5e initial position:**

[image]

**I execute the trajectory:**

`ros2 action send_goal /follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{
  trajectory: {
    joint_names: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint],
    points: [
      { positions: [3.02, -1.63, -1.88, 1.01, 1.51, 1.13], velocities: [0.1, 0.1, 0.1, 0.1, 0.1, 0.1], accelerations: [0.1, 0.1, 0.1, 0.1, 0.1, 0.1], time_from_start: { sec: 5, nanosec: 500 } },      
    ]
  },
  goal_tolerance: [
    { name: shoulder_pan_joint, position: 0.01 },
    { name: shoulder_lift_joint, position: 0.01 },
    { name: elbow_joint, position: 0.01 },
    { name: wrist_1_joint, position: 0.01 },
    { name: wrist_2_joint, position: 0.01 },
    { name: wrist_3_joint, position: 0.01 }
  ]
}"`

**All correct:**

[image]

**I reset the simulation :**

[image]

**I execute the same trajectory :**

[image]

The state is "**Goal Accepted**" in the action_server, but the trajectory never executed.

Thanks in advance for your answer.

DavidMansolino, 2019-10-15:
Hi @LuisYordano, thank you for this precise bug report. I can indeed reproduce this.
In general, when using ROS2 you should not revert or reload the simulation as this will stop the ROS2 node communicating with Webots.
But there is indeed a bug, when the simulation is reverted the ROS2 node does not end correctly (which is why you did get the `Goal Accepted`), I am attempting to fix this in #39 (without success for now).

LuisYordano, 2019-10-15:
Hi @DavidMansolino,

I don't know if this helps, but  in Gazebo there is the commands:

```
- rosservice call /gazebo/reset_simulation "{}"

- rosservice call /gazebo/reset_world "{}"
 
- rosservice call /gazebo/pause_physics "{}"
 
- rosservice call /gazebo/unpause_physics "{}"

```
Webots is different from Gazebo, but the concepts of Reset Simulation, World, Pause, UnPause, would be implemented with the same philosophy in Webots as services in ROS2.

DavidMansolino, 2019-10-15:
Thank you for the information.
We have indeed not yet implemented these functionalities, but we will shortly so that you can easily reset the world.
Just one small question, when you reset the world, the nodes remain connected to Gazebo, right?

LuisYordano, 2019-10-15:
Affirmative!!!
Nodes only terminate communication with ROS Master (ROS1) and ROS2 when the user press CTRL+C.

lukicdarkoo, 2021-09-28:
What about respawning the driver:
```py
tiago_driver = Node(
    package='webots_ros2_driver',
    executable='driver',
    parameters=[
        {'robot_description': robot_description }
    ],

    # Every time one simulation resets the simulation the controller is automatically respawned
   respawn=True
)
```
In addition, you will need some mechanism to reset `ros2_controllers` in case you use them.

lukicdarkoo, 2021-09-28:
I will close the issue. Feel free to re-open it if the answer doesn't solve your problem

closed 2021-09-28 by lukicdarkoo