cyberbotics/webots_ros2 #44: ROS2 Eloquent Release

DavidMansolino, 2019-11-20:
ROS2 Eloquent is going to be released very soon. We should support it as soon as possible:
  - [ ] Support of `control_msgs` on eloquent: ros-controls/control_msgs#37
  - [ ] Extension of the test suite.
  - [ ] New release with ROS2 eloquent.

DavidMansolino, 2019-11-27:
First blocking step almost achieved: ros/rosdistro#23170

DavidMansolino, 2020-06-18:
Eloquent release done:
  - https://github.com/ros/rosdistro/pull/25519

Foxy release done:
  - https://github.com/ros/rosdistro/pull/25521

lukicdarkoo, 2020-07-02:
Maybe we could include `webots_ros2` from `develop` branch to ROS 2 Rolling Ridley:
https://www.ros.org/reps/rep-2002.html

> The Rolling distribution of ROS 2 serves as a staging area for future stable distributions of ROS 2 and as a collection of the most recent development releases.

Here is the relevant `distribution.yaml`:
https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml

DavidMansolino, 2020-07-02:
Yes good idea, but the first thing to do is to enable the tests for rolling ridley

DavidMansolino, 2020-07-10:
Foxy just officially released: https://discourse.ros.org/t/new-packages-and-patch-release-for-ros-2-foxy-fitzroy-2020-07-10/15348

PR #45 (2019-12-06): Add Eloquent to Release
Fix  #44
closed 2020-09-11 by DavidMansolino