cyberbotics/webots_ros2 #60: Create ROS2 reusable nodes

lukicdarkoo, 2020-04-01:
Currently, there is a driver for each robot and coding something like that takes time. It could be improved by creating reusable nodes that can drive different parts of the robot, e.g. there could be a differential drive component that interacts with motors and encoders (Webots), and publishes Odometry and subscribes to Twist topic.

Here is the initial request:
https://github.com/cyberbotics/webots_ros2/pull/59#pullrequestreview-384859672

lukicdarkoo, 2020-04-02:
@DavidMansolino Should a module (e.g. `WebotsDifferentialWheelNode`) be instantiated directly in launch file (e.g. `/webots_ros2_epuck/launch/example_launch.py`) or in a driver (e.g. `/webots_ros2_epuck/webots_ros2_epuck/driver.py`)?

Adding module directly to the launch file may be simpler for users, but then there are two nodes driving the same robot which can make things complicated, e.g. `this.robot.step()` would need to be called from two places. Moreover, the current implementation of `WebotsNode` will not allow two nodes to drive the same robot.

DavidMansolino, 2020-04-02:
Actually what I would do is to make ``WebotsDifferentialWheelNode`` a class that inherits from the ``WebotsNode``class, then the ``webots_ros2_epuck::Driver`` class would inherit from the ``WebotsDifferentialWheelNode`` instead of ``WebotsNode``. Something like this:
```
Driver->WebotsDifferentialWheelNode->WebotsNode
```

lukicdarkoo, 2020-04-02:
Understand, so a user would need to create a driver file.

But is it a good idea to inherit `WebotsDifferentialWheelNode`? Does it scale? If we multiple more modules than we will need to inherit multiple classes. Sometimes people try to avoid this. 

Maybe something like
```
Driver -> WebotsNode
  | -- WebotsDifferentialWheelNode
  | -- WebotsLidarNode
  | -- WebotsRangeNode
  | -- ...
```

Also, I was thinking e.g.  WebotsRangeNode to be able to identify all distance sensors and automatically create corresponding ROS2 topics.

DavidMansolino, 2020-04-02:
You're write, in general it would not scale, but since we are speaking here of the type of robot, this is kind of unique. For example, if in the future we have a ``WebotsArmNode``for a robotic arm then it will never need ``WebotsDifferentialWheelNode`` (same apply for ``WebotsQuadrotorNode`` or so), but for the devices it makes sense, in particular:

>  I was thinking e.g. WebotsRangeNode to be able to identify all distance sensors and automatically create corresponding ROS2 topics.

You're too fast 😄, this was going to be the next things I would suggest, we have for example already this for lidars:

https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_core/webots_ros2_core/laser_publisher.py

But it would be nice to scale this to new devices you are using here (e.g. range, light and camera at least). Ideally I the long term we want this for all the Webots sensors.

lukicdarkoo, 2020-04-02:
Oh, I understand now! I misunderstood the role of `WebotsDifferentialWheelNode` :/

PR #61 (2020-04-06): Add Reusable Differential Drive Node
**Description**
Our objective is to create a reusable component that could be shared within all differential drive robots.

**Related Issues**
Closes #60 

**Affected Packages**
  - webots_ros2_epuck
  - webots_ros2_core
  - webots_ros2_tiago

**Tasks**
  - [x] Create differential drive node
  - [x] Integrate the node into e-puck robot
  - [x] Integrate the node into Tiago robot
  - [x] Create a new RViz config file specifically for Tiago
closed 2020-04-06 by lukicdarkoo