cyberbotics/webots_ros2 #62: Got errors while testing the examples

kaixin-bai, 2020-04-05:
while ros2 launch the universal examples in webots_ros2, i got error like below. And i test is it in anaconda and default python environment, the errors are the same. Could somebody help to fix it?
In Webots the environment runs, but the ur robots didn't move.
```
(base) kb@kb-TM1705:~/ros_projects/ros2_ws$ ros2 launch webots_ros2_universal_robot universal_robot_multiple.launch.py
[INFO] [launch]: All log files can be found below /home/kb/.ros/log/2020-04-05-17-36-18-988216-kb-TM1705-10292
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots_launcher-1]: process started with pid [10302]
[INFO] [universal_robot-2]: process started with pid [10303]
[INFO] [universal_robot-3]: process started with pid [10304]
[universal_robot-2] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 1 second)
[universal_robot-3] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 1 second)
[universal_robot-2] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 2 seconds)
[universal_robot-3] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 2 seconds)
[universal_robot-2] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 3 seconds)
[universal_robot-3] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 3 seconds)
[universal_robot-2] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 4 seconds)
[universal_robot-3] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 4 seconds)
[universal_robot-2] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 5 seconds)
[universal_robot-3] Cannot open file: /tmp/webots-10305/WEBOTS_SERVER (retrying in 5 seconds)
[universal_robot-2] Traceback (most recent call last):
[universal_robot-2]   File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_py/import_type_support_impl.py", line 40, in import_type_support
[universal_robot-2]     return importlib.import_module(module_name, package=pkg_name)
[universal_robot-2]   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
[universal_robot-2]     return _bootstrap._gcd_import(name[level:], package, level)
[universal_robot-2]   File "<frozen importlib._bootstrap>", line 994, in _gcd_import
[universal_robot-2]   File "<frozen importlib._bootstrap>", line 971, in _find_and_load
[universal_robot-2]   File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
[universal_robot-2] ModuleNotFoundError: No module named 'webots_ros2_msgs.webots_ros2_msgs_s__rosidl_typesupport_c'
[universal_robot-2] 
[universal_robot-2] During handling of the above exception, another exception occurred:
[universal_robot-2] 
[universal_robot-2] Traceback (most recent call last):
[universal_robot-2]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot", line 11, in <module>
[universal_robot-2]     load_entry_point('webots-ros2-universal-robot==0.0.2', 'console_scripts', 'universal_robot')()
[universal_robot-2]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 44, in main
[universal_robot-2]     actionServer = ActionServerNode(args=args)
[universal_robot-2]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 29, in __init__
[universal_robot-2]     super().__init__('ur_driver', args=args)
[universal_robot-2]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_core/lib/python3.6/site-packages/webots_ros2_core/webots_node.py", line 57, in __init__
[universal_robot-2]     self.stepService = self.create_service(SetInt, 'step', self.step_callback)
[universal_robot-2]   File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/node.py", line 1219, in create_service
[universal_robot-2]     check_for_type_support(srv_type)
[universal_robot-2]   File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
[universal_robot-2]     msg_type.__class__.__import_type_support__()
[universal_robot-2]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_msgs/lib/python3.7/site-packages/webots_ros2_msgs/srv/_set_int.py", line 256, in __import_type_support__
[universal_robot-2]     module = import_type_support('webots_ros2_msgs')
[universal_robot-2]   File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_py/import_type_support_impl.py", line 42, in import_type_support
[universal_robot-2]     raise UnsupportedTypeSupport(pkg_name)
[universal_robot-2] rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'webots_ros2_msgs'
[universal_robot-3] Traceback (most recent call last):
[universal_robot-3]   File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_py/import_type_support_impl.py", line 40, in import_type_support
[universal_robot-3]     return importlib.import_module(module_name, package=pkg_name)
[universal_robot-3]   File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
[universal_robot-3]     return _bootstrap._gcd_import(name[level:], package, level)
[universal_robot-3]   File "<frozen importlib._bootstrap>", line 994, in _gcd_import
[universal_robot-3]   File "<frozen importlib._bootstrap>", line 971, in _find_and_load
[universal_robot-3]   File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
[universal_robot-3] ModuleNotFoundError: No module named 'webots_ros2_msgs.webots_ros2_msgs_s__rosidl_typesupport_c'
[universal_robot-3] 
[universal_robot-3] During handling of the above exception, another exception occurred:
[universal_robot-3] 
[universal_robot-3] Traceback (most recent call last):
[universal_robot-3]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot", line 11, in <module>
[universal_robot-3]     load_entry_point('webots-ros2-universal-robot==0.0.2', 'console_scripts', 'universal_robot')()
[universal_robot-3]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 44, in main
[universal_robot-3]     actionServer = ActionServerNode(args=args)
[universal_robot-3]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/python3.6/site-packages/webots_ros2_universal_robot/universal_robot.py", line 29, in __init__
[universal_robot-3]     super().__init__('ur_driver', args=args)
[universal_robot-3]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_core/lib/python3.6/site-packages/webots_ros2_core/webots_node.py", line 57, in __init__
[universal_robot-3]     self.stepService = self.create_service(SetInt, 'step', self.step_callback)
[universal_robot-3]   File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/node.py", line 1219, in create_service
[universal_robot-3]     check_for_type_support(srv_type)
[universal_robot-3]   File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
[universal_robot-3]     msg_type.__class__.__import_type_support__()
[universal_robot-3]   File "/home/kb/ros_projects/ros2_ws/install/webots_ros2_msgs/lib/python3.7/site-packages/webots_ros2_msgs/srv/_set_int.py", line 256, in __import_type_support__
[universal_robot-3]     module = import_type_support('webots_ros2_msgs')
[universal_robot-3]   File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_py/import_type_support_impl.py", line 42, in import_type_support
[universal_robot-3]     raise UnsupportedTypeSupport(pkg_name)
[universal_robot-3] rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'webots_ros2_msgs'
[ERROR] [universal_robot-2]: process has died [pid 10303, exit code 1, cmd '/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot --webots-robot-name=UR3e __ns:=/UR3e __params:=/tmp/launch_params_91faa_6u'].
[ERROR] [universal_robot-3]: process has died [pid 10304, exit code 1, cmd '/home/kb/ros_projects/ros2_ws/install/webots_ros2_universal_robot/lib/webots_ros2_universal_robot/universal_robot --webots-robot-name=UR5e __ns:=/UR5e __params:=/tmp/launch_params_5pvlzidx'].
```

kaixin-bai, 2020-04-05:
remove install folder, deactivate conda env and rebuild the package fix it.

closed 2020-04-05 by kaixin-bai

233Xiaowu, 2023-10-27:
hello, I got the same problem as you, but I'm having trouble solving it, could you explain your solution a little more clearly please?
I have delete the deb install bag ，and rebuild it not in the conda environment ，but it did not worked.