cyberbotics/webots_ros2 #65: When launching examples in eloquent I get "[webots_launcher-1] webots_launcher: error: unrecognized arguments: --ros-args"

RDaneelOlivav, 2020-04-14:
**Describe the Bug**
I have installed:
* Webots version: R2020a revision 1
* ROS2 Eloquent

When I launch :
    
> ros2 launch webots_ros2_examples example.launch.py

I get the following error:

```
[INFO] [launch]: All log files can be found below /home/finn/.ros/log/2020-04-14-08-22-05-128650-rdaneelcomputer-1578
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [webots_launcher-1]: process started with pid [1589]
[INFO] [example_controller-2]: process started with pid [1590]
[webots_launcher-1] usage: webots_launcher [-h] [--world WORLD] [--mode MODE] [--no-gui]
[webots_launcher-1] webots_launcher: error: unrecognized arguments: --ros-args
[ERROR] [webots_launcher-1]: process has died [pid 1589, exit code 2, cmd '/home/finn/catkin_ws/install/webots_ros2_core/lib/webots_ros2_core/webots_launcher --mode=realtime --world=/home/finn/catkin_ws/install/webots_ros2_examples/share/webots_ros2_examples/worlds/ros_example.wbt --ros-args'].
[INFO] [example_controller-2]: sending signal 'SIGINT' to process[example_controller-2]
[example_controller-2] Webots doesn't seems to be ready yet: (retrying in 1 second)
[example_controller-2] Webots doesn't seems to be ready yet: (retrying in 2 seconds)
[example_controller-2] Webots doesn't seems to be ready yet: (retrying in 3 seconds)
[ERROR] [example_controller-2]: process[example_controller-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [example_controller-2]: sending signal 'SIGTERM' to process[example_controller-2]
[example_controller-2] Webots doesn't seems to be ready yet: (retrying in 4 seconds)
[ERROR] [example_controller-2]: process has died [pid 1590, exit code -15, cmd '/home/finn/catkin_ws/install/webots_ros2_examples/lib/webots_ros2_examples/example_controller --ros-args --params-file /tmp/launch_params_gyebwfke'].
```

I compiled the webots_ros2 in my catkin_ws without issues, I souced with ```source install/local_setup.bash``` and ros2 works correctly as well as webots. Is this ros-args error that avoid it to be launched because launching:
`
/home/finn/catkin_ws/install/webots_ros2_core/lib/webots_ros2_core/webots_launcher --mode=realtime --world=/home/finn/catkin_ws/install/webots_ros2_examples/share/webots_ros2_examples/worlds/ros_example.wbt`

Has no issue, it starts correctly. 

Any ideas on where it could be the issue? Do I have to set something else?

DavidMansolino, 2020-04-14:
Can you confirm you are compiling from the eloquent branch?

RDaneelOlivav, 2020-04-15:
Confirmed:

finn@rdaneelcomputer:~/catkin_ws/src/webots_ros2$ git branch
* eloquent
  master

DavidMansolino, 2020-04-15:
Ok this can indeed explained what you are experiencing, the eloquent branch was not up to date with master, I did synchronised it with master yesterday.
It should now work, can you try to pull the latest changes and recompile the packages?

RDaneelOlivav, 2020-04-16:
Working ;)

closed 2020-09-02 by DavidMansolino