cyberbotics/webots_ros2 #70: macOS compile error

AndrewJSchoen, 2020-04-22:
**Describe the Bug**
On Mac, where the Webots application is installed in `/Applications`, there is an error when compiling a ROS2 workspace with the webots_ros2_desktop package.

```
>> colcon build --packages-select webots_ros2_desktop
Starting >>> webots_ros2_desktop
--- stderr: webots_ros2_desktop                     
error: can't copy '../../../../../../../Applications/Webots.app/lib/webots/libzip.dylib': doesn't exist or not a regular file
```

**Steps to Reproduce**
1. Install Webots.app from GitHub releases
2. Set `WEBOTS_HOME` to `Applications/Webots.app`
2. Add `webots_ros2` repo (eloquent branch) to `src`
3. colcon build --packages-select webots_ros2_desktop
4. See error

**Expected behavior**
Successful compilation of package.

**Affected Packages**
List of affected packages:
  - webots_ros2_desktop
**System**
 - Webots Version: R2020a Revision 1
 - ROS Version: Eloquent
 - Operating System: macOS Catalina
 - Graphics Card: [AMD Radeon Pro 5300M, Intel UHD Graphics 630]

DavidMansolino, 2020-04-23:
Nice to have some tester on Mac OS too 😃 

First, as a workaround, if you already have Webots installed and WEBOTS_HOME set, you should be able to use all the other packages without the ``webots_ros2_desktop``.

About the compilation issue, can you confirm that you have a ``Applications/Webots.app/lib/webots/libzip.dylib`` file on your system?

AndrewJSchoen, 2020-04-23:
Hi David,

Thanks for the suggestion. I will proceed that way in the meantime. It seems based on the documentation that I just set the robot controller to <extern> and publish PoseTrajectory messages?

With regards to the dylib file, all the files are there, which I found a bit strange. I tried modifying the setup.py file to look in slightly different levels, in case there was an issue in the relative paths, but that didn’t seem to work either.

DavidMansolino, 2020-04-23:
You're welcome,

To use this you need indeed to set the controller of the robot to `<extern>`, then you can use the ros nodes available in these packages, for the ABB and URe robots you will find nodes working out of the box: ``webots_ros2_abb:abb_driver`` and ``webots_ros2_universal_robot::universal_robot``.
You will also find some launch file examples that will launch Webots and the nodes at the same time: https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_abb/launch

If you want to use another robot arm than the URe or ABB ones, you can either use the ABB one which is quite generic (except the name), or use the ABB one as example (it is very simple) to write your own node: https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_abb/webots_ros2_abb/abb_driver.py

AndrewJSchoen, 2020-04-23:
Hmm. I just listed the `lib/webots` directory in the Webots.app package, and get this:

```
>> ls -l Webots.app/lib/webots
total 23208
-rwxr-xr-x@ 1 me  admin   471928 Jan 14 07:30 libOIS.dylib
-rwxr-xr-x@ 1 me  admin  4959580 Jan 14 07:30 libassimp.dylib
-rwxr-xr-x@ 1 me  admin  2018220 Jan 14 07:30 libcrypto.1.0.0.dylib
lrwxr-xr-x  1 schoen  admin       21 Jan 14 07:30 libcrypto.dylib -> libcrypto.1.0.0.dylib
-rwxr-xr-x@ 1 me  admin  1495740 Jan 14 07:30 libgd.3.dylib
-rwxr-xr-x@ 1 me  admin   190912 Jan 14 07:30 liblua.dylib
-rwxr-xr-x@ 1 me  admin  1353976 Jan 14 07:30 libode.dylib
-rwxr-xr-x@ 1 me  admin   480768 Jan 14 07:30 libopenal.dylib
-rw-r--r--@ 1 me  admin   475716 Jan 14 07:30 libpico.dylib
lrwxr-xr-x  1 me  admin       14 Jan 14 07:30 libssh.dylib -> libssh.4.dylib
-rwxr-xr-x@ 1 me  admin   414512 Jan 14 07:30 libssl.1.0.0.dylib
lrwxr-xr-x  1 me  admin       18 Jan 14 07:30 libssl.dylib -> libssl.1.0.0.dylib
lrwxr-xr-x  1 me  admin       14 Jan 14 07:30 libzip.dylib -> libzip.2.dylib
drwxr-xr-x@ 4 me  admin      128 Jan 14 07:30 qt
```

I'm not seeing `libzip.2.dylib`, but I wonder if that being missing could be part of the issue? Let me know if there is any other info that would be helpful and I can try to get it/try things.

DavidMansolino, 2020-04-23:
That my be the problem indeed, it looks like libzip.dylib is just a symlink to libzip.2.dylib, but if libzip.2.dylib doesn't exist it might indeed cause issues.
Which version of Webots are you using?

AndrewJSchoen, 2020-04-23:
Should be `webots-R2020a-rev1`

DavidMansolino, 2020-04-23:
Ok, may I ask you to try with a beta of R2020a-rev2 that you can download here:https://github.com/cyberbotics/webots/releases
We have cleaned/removed many dependencies (libzip for example), that might solve the issue.

AndrewJSchoen, 2020-04-23:
Sounds good. I’ll give that a try!

zmk5, 2020-08-04:
I had the package not able to install until I set `export WEBOTS_HOME=Applications/Webots.app`. Once I did that I was able to get it to install on macOS Catalina. I'll write down the things I've done so maybe to update the installation guide for macOS if that is ok with you guys.

zmk5, 2020-08-04:
I'm running into the same problem. I installed R2020b and I run into this error when running `colcon build`

```bash
FileNotFoundError: [Errno 2] No such file or directory: '/Users/zmk5/ROS2/webots_ws/src/webots_ros2/webots_ros2_desktop/../../../../../../../Applications/Webots.app/projects/default/controllers/ros/include/boost/tti/detail/dtype.hpp' -> '/Users/zmk5/ROS2/webots_ws/build/webots_ros2_desktop/../../../../../../../Applications/Webots.app/projects/default/controllers/ros/include/boost/tti/detail/dtype.hpp'
```

zmk5, 2020-08-04:
Hey, I followed the above with webots 2020b and I come upon this error:

```
FileNotFoundError: [Errno 2] No such file or directory: '/Users/zmk5/ROS2/webots_ws/build/webots_ros2_desktop/../../../../../../../Applications/Webots.app/projects/default/controllers/ros/include/boost/mpl/aux_/preprocessed/gcc'
```

When I looked into the installation path I end up with this:

```bash
~$ zmk5$ ls /Applications/Webots.app/projects/default/controllers/ros/include/boost/mpl/aux_/preprocessed/
bcc/        bcc_pre590/  gcc/        msvc70/     no_ctps/    plain/
bcc551/     dmc/        msvc60/     mwcw/       no_ttp/
```

`gcc` shows up as a directory and not a file.

closed 2020-08-06 by DavidMansolino