cyberbotics/webots_ros2 #71: Improper conversion to rclpy.Time for nanoseconds

3wnbr1, 2020-04-23:
In `webots_node`, time is converted from unix timestamp format to `rclpy.Time`

https://github.com/cyberbotics/webots_ros2/blob/40385d93f49812357145cb9b285758b8fcbcbf55/webots_ros2_core/webots_ros2_core/webots_node.py#L93

However, there seems to be an error in the conversion as stated nanoseconds are in fact microseconds. Therefore, the line should be :

```python
nanoseconds = int((sim_time - seconds) * 1e9)
```

DavidMansolino, 2020-04-23:
You are completely right! We are going to fix this soon.

closed 2020-04-24 by lukicdarkoo

lukicdarkoo, 2020-04-24:
@3wnbr1 Thank you for reporting the issue!

We have decided to delete `now()` method in favor of [`self.get_clock().now().to_msg()`]( http://docs.ros2.org/eloquent/api/rclpy/api/node.html?highlight=get_clock#rclpy.node.Node.get_clock). It is ROS2 way of dealing with time:
- if `--ros-args --param use_sim_time:=true` is used, your node will use simulated time,
- otherwise it will automatically switch to real-time.

If you exclusively want to use the simulated time you can use:
```
Time(seconds=self.robot.getTime()).to_msg()
```

PR #74 (2020-04-24): Replace .now() method
**Description**
Nanoseconds are not calculated properly.

**Related Issues**
Fixes #71

**Affected Packages**
  - webots_ros2_core
  - webots_ros2_epuck
  - webots_ros2_demos

**Tasks**
  - [x] Decide wether to deprecate `.now()` method in favor of `self.get_clock().now().to_msg()` or to fix it
  - [x] Fix time usage in:
    - [x] `driver.py` (epuck), 
    - [x] `trajectory_follower.py`,
    - [x] `follow_joint_trajectory_client.py`,
    - [x] `tf_publisher.py` and
    - [x] `laser_publisher.py` (not tested)
  - [x] Increase line length in `webots_ros2_core`

**Additional [... cut]