cyberbotics/webots_ros2 #79: Synchronisation parameter declaration error for WebotsNode on ROS 2 Foxy

3wnbr1, 2020-05-09:
**Describe the Bug**
Using ROS 2 Foxy (master), any `WebotsNode` crashes on `__init__` when trying to declare `synchronisation` parameter.

**Steps to Reproduce**
1. Use ROS 2 foxy (master) built from source
2. Build webots_ros2 from master
3. Run any example/demo code using `WebotsNode` base node from `webots_ros2_core`

**Expected behavior**
Parameter `synchronisation` must be declared successfully

**Affected Packages**
List of affected packages:
  - webots_ros2_core

**Stracktrace**
```
Traceback (most recent call last):
  File "/home/user/ros/ros/install/test_package/lib/test_package/test_controller", line 11, in <module>
    load_entry_point('test_package', 'console_scripts', 'test_controller')()
  File "/home/user/ros/ros/build/test_package/test_package/test_controller.py", line 36, in main
    test_controller = TestController(args=args)
  File "/home/user/ros/ros/build/test_package/test_package/test_controller.py", line 21, in __init__
    super().__init__('test_controller', args)
  File "/home/user/ros_ws/build/webots_ros2_core/webots_ros2_core/webots_node.py", line 44, in __init__
    self.declare_parameter('synchronization',
  File "/home/user/ros2_foxy/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 343, in declare_parameter
    return self.declare_parameters('', [(name, value, descriptor)], ignore_override)[0]
  File "/home/user/ros2_foxy/install/rclpy/lib/python3.8/site-packages/rclpy/node.py", line 438, in declare_parameters
    parameter_list.append(Parameter(name, value=value))
  File "/home/user/ros2_foxy/install/rclpy/lib/python3.8/site-packages/rclpy/parameter.py", line 127, in __init__
    type_ = Parameter.Type.from_parameter_value(value)
  File "/home/user/ros2_foxy/install/rclpy/lib/python3.8/site-packages/rclpy/parameter.py", line 70, in from_parameter_value
    raise TypeError('The given value is not one of the allowed types.')
TypeError: The given value is not one of the allowed types.
```

**System**
 - Webots Version: R2020a revision 1
 - ROS Version: Foxy (master)
 - Operating System: Ubuntu Focal Fossa (20.04LTS) / macOS 10.15.4
 - Graphics Card: N/A
 - Python : Python 3.8.2

**Additional context**
Parameter declaration is no longer compatible with ros foxy

https://github.com/cyberbotics/webots_ros2/blob/a2251d475e45f83be99ba803a531ce8425d25e80/webots_ros2_core/webots_ros2_core/webots_node.py#L44-L45

The following line is effective
```python
self.declare_parameter('synchronization', False)
```

DavidMansolino, 2020-05-11:
Thank you for this bug report, we have indeed not tested the packages with ROS Foxy yet, we will fix this!

3wnbr1, 2020-05-13:
I tested today `master` branch against `eloquent` Debian released packages and got the same backtrace :

```
Traceback (most recent call last):
  File "/home/runner/work/ros/ros/install/test_package/lib/test_package/test_controller", line 11, in <module>
    load_entry_point('test_package', 'console_scripts', 'test_controller')()
  File "/home/runner/work/ros/ros/build/test_package/test_package/test_controller.py", line 36, in main
    test_controller = TestController(args=args)
  File "/home/runner/work/ros/ros/build/test_package/test_package/test_controller.py", line 21, in __init__
    super().__init__('test_controller', args)
  File "/home/runner/ros_ws/build/webots_ros2_core/webots_ros2_core/webots_node.py", line 45, in __init__
    Parameter('synchronization', Parameter.Type.BOOL, False))
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/node.py", line 333, in declare_parameter
    return self.declare_parameters('', [(name, value, descriptor)], ignore_override)[0]
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/node.py", line 428, in declare_parameters
    parameter_list.append(Parameter(name, value=value))
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/parameter.py", line 127, in __init__
    type_ = Parameter.Type.from_parameter_value(value)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/parameter.py", line 70, in from_parameter_value
    raise TypeError('The given value is not one of the allowed types.')
```
The `foxy` parameter declaration fix also works fine on `eloquent`

PR #80 (2020-05-13): Fix Foxy Parameter Compatibility
**Description**
The following line is not compatible with Foxy:
https://github.com/cyberbotics/webots_ros2/blob/a2251d475e45f83be99ba803a531ce8425d25e80/webots_ros2_core/webots_ros2_core/webots_node.py#L44-L45

**Related Issues**
Closes #79

**Affected Packages**
  - webots_ros2_core

**Tasks**
  - [x] Fix compatibility
  - [x] Verify if `get_parameter_or()` is affected (https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/webots_ros2_universal_robot/universal_robot.py#L30 and
https://github.com/cyberbotics/webots_ros2/blob/master/ [... cut]
closed 2020-05-13 by lukicdarkoo