cyberbotics/webots_ros2 #112: Is possible implement webots_ros2 as c++ programme

mrliujie, 2020-07-07:
Hello, thanks for such a nice project.

Thanks for such a nice project. I want to implement my webots_ros2 project using c++ because the original webots_ros2 is implemented by python. 
I want to implement an external controller using directly c++ API and publish the simulation data with ROS2. And now i can comunicate with webots using extern controller.
And now I can communicate with webots using an extern controller.

My question is :
1. Is it wise to write the project without using the provided python project?
2. In webots/project/default/controllers/ros, is it necessary to implement the class named "RosSupervisor", what is RosSupervisor use for?

DavidMansolino, 2020-07-07:
Hi,

Yes, this is for sure possible to create a C++ interface instead of Python, we decided to use python for simplicity of deployment and release (simpler dependencies management, cross-platform, etc.).

However, why would you need to create another interface? can't you use the Python existing one? You can in any case write your control node in the language you want.

About `webots/project/default/controllers/ros` this is the ros1 interface, it works completely differently, you should rather take inspiration from the python version from this repo (the ROS2 interface is much more clean, stable and scalable).

mrliujie, 2020-07-13:
Thanks a lot. I think you're right. I will use the Python existing one.

closed 2020-07-13 by mrliujie

DavidMansolino, 2020-07-13:
You're welcome :wink: