cyberbotics/webots_ros2 #121: No rosdep command in windows10

dzywater, 2020-08-06:
[image]

System: ros2 foxy, windows 10 19041

Build failed!
I didn’t run: rosdep update
Because no this command. How to install rosdep.

DavidMansolino, 2020-08-06:
ROS2 on Windows is still very experimental, this issue is not directly linked to the webots_ros2 packages.

closed 2020-08-06 by omichel

dzywater, 2020-08-06:
> ROS2 on Windows is still very experimental, this issue is not directly linked to the webots_ros2 packages.

But why does the readme show installation of webots-ros2 based on ros2 on windows 10

omichel, 2020-08-06:
Can you give the URL of this README file?

dzywater, 2020-08-07:
> Can you give the URL of this README file?

Not readme but wiki. As the above picture shows, how to call rosdep update in windows 10?
[image]

omichel, 2020-08-07:
Sorry, I need a URL for this, not a screenshot.

DavidMansolino, 2020-08-07:
This is probably the link: http://wiki.ros.org/webots_ros2#webots_ros2_Package

lukicdarkoo, 2020-08-07:
@dzywater probably didn't clone the submodules:
```bash
git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2.git
```

We have to fix this on ROS Wiki, and this may be something we need to emphasize in this repo in the README files.

omichel, 2020-08-07:
> We have to fix this in the ROS Wiki, and this may be something we need to emphasize in this repo in the README files.

@lukicdarkoo: can you take care of this?

DavidMansolino, 2020-08-07:
FYI, currently I have no more right access anymore to it, I have requested right access again.

lukicdarkoo, 2020-08-07:
@omichel @DavidMansolino 

I am surprised that I have the right to edit ROS Wiki, changed! I will also add to the repo's README in the next PR

dzywater, 2020-08-07:
> @omichel @DavidMansolino
> 
> I am surprised that I have the right to edit ROS Wiki (I thought I am not allowed to change it), changed! I will also add to the repo's README in the next PR

Thanks all. It works with ros2 and windows 10
[image]

james-yoo, 2020-08-14:
@dzywater In my case, 'rosdep update' caused following error:
----------------------------------------------------------------------------------
Add distro "dashing"
ERROR: Rosdep experienced an error: module 'os' has no attribute 'geteuid'
-----------------------------------------------------------------------------------
Do you know how to solve it?