cyberbotics/webots_ros2 #122: install/webots_ros2_desktop has same files as webots

dzywater, 2020-08-08:
I built webots_ros2 in windows10. I found that install/webots_ros2_desktop has same files as webots. Their files are almost same. These files occupy too much capacity.

lukicdarkoo, 2020-08-08:
Check section 2.1.1 at http://wiki.ros.org/webots_ros2

If you don't want to use `webots_ros2_desktop` simply skip it:
```
colcon build --packages-skip webots_ros2_desktop
```
and point `ROS2_WEBOTS_HOME` environment variable to your Webots installation directory.

dzywater, 2020-08-09:
> Solved

closed 2020-08-09 by dzywater

dzywater, 2020-08-09:
> Check section 2.1.1 at http://wiki.ros.org/webots_ros2
> 
> If you don't want to use `webots_ros2_desktop` simply skip it:
> 
> ```
> colcon build --packages-skip webots_ros2_desktop
> ```
> 
> and point `ROS2_WEBOTS_HOME` environment variable to your Webots installation directory.

Thanks! 

I found only python codes are provided. Could you give some c or c++ demos to show how to use webots_ros2!

dzywater, 2020-08-09:
> You can only use Python to write ROS2 driver for Webots (a node that closely interacts with Webots). You can check a very simple example from [webots_ros2_examples](https://github.com/cyberbotics/webots_ros2/blob/foxy/webots_ros2_examples/webots_ros2_examples/example_controller.py) or a bit more complex from [webots_ros2_epuck](https://github.com/cyberbotics/webots_ros2/blob/foxy/webots_ros2_epuck/webots_ros2_epuck/driver.py).
> 
> Soon, we are planning to release a new version of `webots_ros2` that will automatically expose the ROS2 interface based on a model of your robot, with no or minimal coding required in Python:
> https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_core#universal-launcher
> 
> In both cases, only the ROS2 driver for Webots is written in Python. Other ROS2 nodes can, of course, be written in C, or C++, or any other programming language.
> 
> P.S. It is possible to write the ROS2 driver for Webots in C/C++ as well, but we avoid recommending it. Can you tell us your use-case? Can you justify using C/C++ over Python?

I prefer to use c++. Others may not.
Besides, written by c/c++, it is for both python and c/c++, just like webots self. 
It is my suggestion.

dzywater, 2021-08-20:
> You can only use Python to write ROS2 driver for Webots (a node that closely interacts with Webots). You can check a very simple example from [...]

#269 It means I can code by C/C++?

dzywater, 2021-08-20:
I saw the webots_ros2_driver sources. Ha, Ha, Ha, for my preference, I writen a simple version which is similar serveral mouths ago. I feel upset to not maintain it for your updation. I am free now.

lukicdarkoo, 2021-08-20:
Feel free to contribute! I know you primarily use Windows so help from that side is welcome.

The new version supports C++ and Python plugins. It integrates ros2_control, we are running runtime tests in CI, and we are converting all examples to use the new interface.