cyberbotics/webots_ros2 #138: Extend Robot Support

DavidMansolino, 2020-09-11:
The next objective in term of supported robot types (from the [ROSIN project](https://www.rosin-project.eu/ftp/ros2-for-the-e-puck2-robot)] goals) are:
  - [x] Differential-wheel robot: Generic node.
    - [x] E-puck
    - [x] Tiago
  - [x] Robotic arms: Generic node.
    - [x] Ur3/5/10e robotic arms.
    - [x] ABB robotic arms.
    - [ ] 3 other examples to add.
  - [ ] Four wheel vehicle: we need to create a generic node using the Car/Driver API (probably using this kind of messages: http://docs.ros.org/jade/api/ackermann_msgs/html/index-msg.html)
  - [ ] Flying robots: we need to create a generic drone node (probably respecting this REP: https://www.ros.org/reps/rep-0147.html).
    - [ ] DJI' Mavic 2 PRO.
    - [ ] 1 other drone.

lukicdarkoo, 2020-09-28:
We should also integrate existing ROS2 packages:
- Robotic arms -> MoveIt 2,
- Vehicles - > Autoware,
- Flying robots -> maybe PX4 or ArduPilot

lukicdarkoo, 2021-07-22:
The ROSIN project has finished and the missing simulations are largely implemented in the following packages (see `webots_ros2_mavic` and `webots_ros2_tesla`)

closed 2021-07-22 by lukicdarkoo