cyberbotics/webots_ros2 #144: Version 1.0.1, launch error in Ubuntu and windows10

dzywater, 2020-09-21:
[image]
[image]

DavidMansolino, 2020-09-21:
Where is Webots installed on your system? Is WEBOTS_HOME set?

dzywater, 2020-09-22:
> Where is Webots installed on your system? Is WEBOTS_HOME set?

Yes. Both WEBOTS_HOME and ROS_WEBOTS_HOME are set.

dzywater, 2020-09-22:
> Where is Webots installed on your system? Is WEBOTS_HOME set?

Version 1.0.0 are OK. And I compiled version 1.0.0 with option    —packages-skip webots_ros2_desktop

omichel, 2020-09-22:
Which version of Python do you have on Linux?

dzywater, 2020-09-22:
> Which version of Python do you have on Ubuntu20.02-python3.8.2, windows10-python3.8.4

DavidMansolino, 2020-09-22:
Just for the note:

> I compiled version 1.0.0 with option —packages-skip webots_ros2_desktop

This is now useless with version 1.0.1 as this package does not exists anymore.

About your issue, it would be useful if you could check in the log if it says something about which line is causing this issue.

dzywater, 2020-09-22:
> Just for the note:
>
>
> This is now useless with version 1.0.1 as this package does not exists anymore.
>
> About your issue, it would be useful if you could check in the log if it says something about [...]

No useless lines are found 
[image]

dzywater, 2020-09-22:
> Just for the note:
>
>
> This is now useless with version 1.0.1 as this package does not exists anymore.
>
> About your issue, it would be useful if you could check in the log if it says something about [...]

No useful lines are found
[image]

dzywater, 2020-09-22:
> Just for the note:
>
>
> This is now useless with version 1.0.1 as this package does not exists anymore.
>
> About your issue, it would be useful if you could check in the log if it says something about [...]
I try other examples and ‘can’t mix strings and bytes in path components’ always are printed.

lukicdarkoo, 2020-09-24:
Please check if the following works for you:
https://github.com/cyberbotics/webots_ros2/issues/145#issuecomment-698236822

PR #146 (2020-09-25): Decode path from `which` command
**Description**
This PR fixes `Can't mix strings and bytes in path components` by decoding path from `which` command.
It also introduces minor changes to avoid potential recursions (`get_webots_version` <-> `get_webots_home`).

**Related Issues**
Fixes #144, fixes #145

**Affected Packages**
  - webots_ros2_core

**Tasks**
  - [x] Do comprehensive tests to avoid side-effects
closed 2020-09-25 by lukicdarkoo