cyberbotics/webots_ros2 #145: robot_state_publisher failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'

Drojas251, 2020-09-21:
I am trying to run the webots_ros2_universal_robots package though running into, what seems to be, a robot_state_publisher error. I am using Ubuntu 20.04 LTS running ROS2 Foxy Desktop. I installed webots from Debian Packages with APT and installed the webots_ros2 packages from source [following these instructions](http://wiki.ros.org/webots_ros2) . 

When running the following example [ros2 launch webots_ros2_universal_robot universal_robot.launch.py] I get the error which I posted below. I went though the files and made the python scripts executables [ chmod +x filename.py], rebuilt my workspace, though the same error occurs. 

Prior to this, I installed webots Debian Packages with APT and webots_ros2 packages also with APT. Everything was working fine, and I was able to run the webot_ros2_universal_robot demos. I needed to modify the webot_ros2_universal_robot packages, so I created a workspace and installed them from source which is when I first ran into the error. I uninstalled the previous webot_ros2 packages installed with APT, thinking there were conflicting file paths, though that did not fix the error. I even created a new Virtual environment and installed webots debian packages and webots_ros2 from source. However the same error is present. 

Any help would be greatly appreciated. 

**ERROR MESSAGE** 

iego@diego-virtual-machine:~$ source /opt/ros/foxy/setup.bash 
diego@diego-virtual-machine:~$ cd ros2_ws/
diego@diego-virtual-machine:~/ros2_ws$ source install/setup.bash 
diego@diego-virtual-machine:~/ros2_ws$ ros2 launch webots_ros2_universal_robot universal_robot.launch.py
[INFO] [launch]: All log files can be found below /home/diego/.ros/log/2020-09-21-12-15-48-999896-diego-virtual-machine-2264
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=TypeError("Can't mix strings and bytes in path components")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 775, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 635, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 635, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 184, in perform_substitution
    return substitution.perform(self)
  File "/home/diego/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/webots_launcher.py", line 98, in perform
    webots_path = get_webots_home()
  File "/home/diego/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/utils.py", line 59, in get_webots_home
    if os.path.isdir(path) and get_webots_version(path) == version:
  File "/home/diego/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/utils.py", line 140, in get_webots_version
    versionFile = os.path.join(path if path is not None else get_webots_home(),
  File "/usr/lib/python3.8/posixpath.py", line 90, in join
    genericpath._check_arg_types('join', a, *p)
  File "/usr/lib/python3.8/genericpath.py", line 155, in _check_arg_types
    raise TypeError("Can't mix strings and bytes in path components") from None
TypeError: Can't mix strings and bytes in path components
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher-1]: process started with pid [2286]
**[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] [INFO] [1600715749.248895892] [robot_state_publisher]: got segment 
[ERROR] [robot_state_publisher-1]: process[robot_state_publisher-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher-1]: sending signal 'SIGTERM' to process[robot_state_publisher-1]
[ERROR] [robot_state_publisher-1]: process has died [pid 2286, exit code -15, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_dcugksof'].**

lukicdarkoo, 2020-09-24:
I am trying to reproduce the error. Which branch did you use, `master`, `foxy`, or `develop`?

Drojas251, 2020-09-24:
I used foxy branch 

Installed from sources by following [this](http://wiki.ros.org/webots_ros2)

    source /opt/ros/$ROS_DISTRO/setup.bash
    
    # Retrieve the sources
    cd /path/to/ros2_ws/src
    git clone --recurse-submodules -b foxy https://github.com/cyberbotics/webots_ros2.git
    cd ..
    
    # Check dependencies
    rosdep update
   rosdep install --from-paths src --ignore-src --rosdistro foxy
   
   # Building packages
   colcon build
 
   # Source this workspace (careful when also sourcing others)
   source install/setup.bash

Drojas251, 2020-09-24:
I get the same error when using the master branch

lukicdarkoo, 2020-09-24:
Good news, I managed to reproduce the issue on another machine. It seems to to be related to #144. Working on it...

lukicdarkoo, 2020-09-24:
I added a fix #146 

Please try it by cloning `fix-which-webots` branch (instead of `foxy`):
```
git clone --recurse-submodules -b fix-which-webots https://github.com/cyberbotics/webots_ros2.git
```

P.S. Don't forget to install `transforms3d`, `pip3 install transforms3d`

Drojas251, 2020-09-24:
Fantastic. This branch is working on my machine. I tried the multiple examples and they are all passing. Thank you

lukicdarkoo, 2020-09-25:
@Drojas251 I am glad it works for you! Thanks for reporting the issue.

PR #146 (2020-09-25): Decode path from `which` command
**Description**
This PR fixes `Can't mix strings and bytes in path components` by decoding path from `which` command.
It also introduces minor changes to avoid potential recursions (`get_webots_version` <-> `get_webots_home`).

**Related Issues**
Fixes #144, fixes #145

**Affected Packages**
  - webots_ros2_core

**Tasks**
  - [x] Do comprehensive tests to avoid side-effects
closed 2020-09-25 by lukicdarkoo