cyberbotics/webots_ros2 #147: Improve the main documentation

lukicdarkoo, 2020-09-30:
**Documentation page(s) concerned**
https://github.com/cyberbotics/webots_ros2/tree/master/webots_ros2_core

**What is the problem with the current documentation?**
It is an important page and, therefore, it should be cleaner.

**Describe how you would fix it**
 A table of content should be added, each code snipped should have a title (so a user can jump to the content directly), and page should be reorganization in general.

DavidMansolino, 2020-10-06:
We should also update this wiki: http://wiki.ros.org/webots_ros2 to point to the doc of this repo.

lukicdarkoo, 2020-10-06:
Documentation rules for ROS2 are on the way:
```
https://github.com/ros2/design/pull/301/files
```
we should keep eyes on that PR.

lukicdarkoo, 2020-10-21:
Done in #161

closed 2020-10-21 by lukicdarkoo