cyberbotics/webots_ros2 #169: Webots running native in Windows connecting to ROS2 running on Docker for Windows.

mrbhjv, 2020-11-28:
## Escenario

- With WSL2 (or just Windows), run a docker (Docker for Windows) container with ros:foxy as the base image, with webots_ros2, installed, launching one of the Webots drivers (or my custom).

- In Windows (not via container) run Webots.

**Expected behavior** the controller running in the container can interact with the Webots instance running locally in Windows.

I've installed Webots on a Windows computer, and I'd like to run my ROS stack, including this package, with Docker. I'd like to avoid having to install ROS2 in Windows, as I can fully work with it via containers.

Following a bit on the workflow of the package, it'd seem that it will always try to start Webots, would it be possible to hook up to a running Webots instance?

---

We are going to host Webots on a Linux machine for CI purposes, however, for development purposes, we're running Windows on our workstations. So, any assistance on how we could solve this problem would be greatly welcome.

Thanks

lukicdarkoo, 2020-11-30:
We have never tried it, but here are a few tips.

Webots uses pipes to communicate with a controller. [Mounting correct files](https://docs.docker.com/storage/volumes/) (search the following files `/tmp/webots*`) when running the Docker container should allow communication between Webots and the controller. In addition, Webots uses shared memory for images and point clouds, so you should also need to make sure [the memory between Docker and the host is shared](https://docs.docker.com/engine/reference/run/#ipc-settings---ipc).

Let us know about your progress and whether we can help you.

mrbhjv, 2020-12-01:
A little update regarding this.

My approach was to first, add another launch configuration, to avoid launching Webots, as I already have it running and webots_ros2_core robot_launch.py will always try to start it, which is not the behavior I want. Next, I made sure my container has access to my Windows installed Webots path, and set the WEBOTS_HOME, thus far, all was good. However, when I launch the ROS process, I got a problem with Python not being able to find a `_controller` module.
This error is because it's not intended to be used on cross platforms (which I knew), and, as the container is running a Linux distro, is trying to go through the Linux way on a Windows install, looking for the .so files (that go along the controller library) in Linux, instead of picking up DLLs.

I did try to add manually the .so files (from a Linux install) to just see if I could make it continue, making it further, however, I later run into an error of `AttributeError: module '_controller' has no attribute 'Joystick_swiginit'` which is when I call it done, as I don't want to keep randomly adding things hoping it works, as I know I'll hit a wall sooner rather than later :).

I don't think that there's a simple solution as far as I can tell, I guess I could run it headless from the WSL, but it's not what I wanted.

Any suggestions will be highly appreciated, as I'm really keen on using this simulator. However, I'm also keen on cross-platform work with Docker.

lukicdarkoo, 2020-12-01:
Unfortunately, I don't believe that approach can work. As you said, you trying to run a Windows compiled Python library in Ubuntu Docker.

[image]
I would install the same Webots version in the Docker image and on the host OS (Windows). The goal would be to use `Webots API` (see the picture) in the Docker image and `Webots Simulation` in Windows. Therefore, you will need to do as stated in my previous comment and to delete `WebotsLauncher` from `robot_launch.py` (as it is not needed anymore in the container):  
https://github.com/cyberbotics/webots_ros2/blob/a58e68837af8b82bcf02a49ed026ae2f5604472f/webots_ros2_core/launch/robot_launch.py#L101 

Alternatively, you can use Webots Docker image and X11 forwarding:
https://www.cyberbotics.com/doc/guide/installation-procedure#run-webots-in-docker-with-gui
It should be also possible to configure X11 forwarding with Windows:
https://medium.com/@potatowagon/how-to-use-gui-apps-in-linux-docker-container-from-windows-host-485d3e1c64a3

mrbhjv, 2021-02-18:
I'm closing this issue as in the end we decided to go to Linux. There are too many problems to work with Docker for Windows. Thanks anyways :)

closed 2021-02-18 by mrbhjv