cyberbotics/webots_ros2 #175: Generate UR5e PROTO from XARCO (instead of independent URDF and PROTO)

lukicdarkoo, 2020-12-01:
This is related to the following discussion:
https://github.com/cyberbotics/webots_ros2/pull/174#issuecomment-736346858

ygoumaz, 2023-02-21:
This seems to be solved but not completely. In [multirobot_launch.py](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/launch/multirobot_launch.py), the UR5e is spawned from the xacro, but in [robot_nodes_launch.py](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/launch/robot_launch/robot_nodes_launch.py), it seems still dependent on the URDF file `ur5e_with_gripper.urdf`.

ygoumaz, 2023-05-03:
The goal is to drop the importer in the future and focus on PROTOs that can be spawned using the supervisor. URDFs and Xacro files should be converted manually with `urdf2webots` and integrated as PROTOs in projects.

closed 2023-05-03 by ygoumaz