cyberbotics/webots_ros2 #180: webots_ros2_turtlebot3 warning: Exporting 'BallJoint' nodes to URDF is currently not supported && he mass ratio between the heaviest solid and the lightest solid  is huge

dzywater, 2020-12-16:
WARNING: It is not recommended to run Webots as root.
WARNING: The mass ratio between the heaviest solid ('Solid' - mass = 95.624 kg) and the lightest solid ('Solid' - mass = 0.000268083 kg) is huge. This is a source of physics instabilities.
WARNING: Fridge (PROTO) > HingeJoint > Solid: heaviest solid
WARNING: TurtleBot3Burger_enu (PROTO) > DEF CASTER_JOINT BallJoint > Solid: lightest solid
INFO: Starting extern controller for robot "TurtleBot3Burger".
WARNING: TurtleBot3Burger_enu (PROTO) > DEF CASTER_JOINT BallJoint: Exporting 'BallJoint' nodes to URDF is currently not supported

lukicdarkoo, 2020-12-16:
```
WARNING: The mass ratio between the heaviest solid ('Solid' - mass = 95.624 kg) and the lightest solid ('Solid' - mass = 0.000268083 kg) is huge. This is a source of physics instabilities.
WARNING: Fridge (PROTO) > HingeJoint > Solid: heaviest solid
WARNING: TurtleBot3Burger_enu (PROTO) > DEF CASTER_JOINT BallJoint > Solid: lightest solid
```
If you update Webots, you should not see this error anymore.

```
WARNING: TurtleBot3Burger_enu (PROTO) > DEF CASTER_JOINT BallJoint: Exporting 'BallJoint' nodes to URDF is currently not supported
```
URDF exporter currently doesn't support ball joints. However, since the ball joint in TurtleBot3 is not motorized (and there is no encoder) it doesn't affect the simulation. You can safely ignore it.

```
WARNING: It is not recommended to run Webots as root.
```
Don't run Webots as a root user :)

closed 2020-12-17 by lukicdarkoo

dzywater, 2020-12-21:
I updated webots to 2021a, then could not run.

D:\>ros2 launch webots_ros2_core robot_launch.py world:=D:\app\ros2ex\out\install\share\webots_ros2_epuck\worlds\epuck_world.wbt
[INFO] [launch]: All log files can be found below C:\Users\Administrator\.ros/log\2020-12-21-10-20-19-847299-DESKTOP-UT0NVAN-7584
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at D:\app\ros2\Lib\site-packages\launch\launch_service.py:271> exception=FileNotFoundError(2, 'The system cannot find the specified file。', None, 2, None)>
Traceback (most recent call last):
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "D:\app\ros2\Lib\site-packages\launch\action.py", line 108, in visit
    return self.execute(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 775, in execute
    self.__expand_substitutions(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 635, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 635, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "D:\app\ros2\Lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "D:\app\ros2\Lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "D:\app\ros2\Lib\site-packages\launch\launch_context.py", line 184, in perform_substitution
    return substitution.perform(self)
  File "D:\app\ros2ex\out\install\Lib\site-packages\webots_ros2_core\webots_launcher.py", line 98, in perform
    webots_path = get_webots_home()
  File "D:\app\ros2ex\out\install\Lib\site-packages\webots_ros2_core\utils.py", line 59, in get_webots_home
    path = os.path.split(os.path.abspath(subprocess.check_output(['which', 'webots'])))[0]
  File "c:\python38\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "c:\python38\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "c:\python38\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\python38\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the specified file。
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at D:\app\ros2\Lib\site-packages\launch\launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "D:\app\ros2\Lib\site-packages\launch\action.py", line 108, in visit
    return self.execute(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher.EXE-1]: process started with pid [1676]
[robot_state_publisher.EXE-1] Parsing robot urdf xml string.
[... cut ...] nt
[ERROR] [robot_state_publisher.EXE-1]: process[robot_state_publisher.EXE-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher.EXE-1]: sending signal 'SIGTERM' to process[robot_state_publisher.EXE-1]
[ERROR] [robot_state_publisher.EXE-1]: process has died [pid 1676, exit code 1, cmd 'D:\app\ros2\lib\robot_state_publisher\robot_state_publisher.EXE --ros-args --params-file C:\Users\ADMINI~1\AppData\Local\Temp\launch_params_tq0uri5w'].

dzywater, 2020-12-21:
I updated webots to 2021a. then could not run.

D:\>ros2 launch webots_ros2_epuck robot_launch.py
[INFO] [launch]: All log files can be found below C:\Users\Administrator\.ros/log\2020-12-21-10-25-41-808335-DESKTOP-UT0NVAN-8424
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at D:\app\ros2\Lib\site-packages\launch\launch_service.py:271> exception=FileNotFoundError(2, 'The system cannot find the specified file。', None, 2, None)>
Traceback (most recent call last):
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "D:\app\ros2\Lib\site-packages\launch\action.py", line 108, in visit
    return self.execute(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 775, in execute
    self.__expand_substitutions(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 635, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 635, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "D:\app\ros2\Lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "D:\app\ros2\Lib\site-packages\launch\utilities\perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "D:\app\ros2\Lib\site-packages\launch\launch_context.py", line 184, in perform_substitution
    return substitution.perform(self)
  File "D:\app\ros2ex\out\install\Lib\site-packages\webots_ros2_core\webots_launcher.py", line 98, in perform
    webots_path = get_webots_home()
  File "D:\app\ros2ex\out\install\Lib\site-packages\webots_ros2_core\utils.py", line 59, in get_webots_home
    path = os.path.split(os.path.abspath(subprocess.check_output(['which', 'webots'])))[0]
  File "c:\python38\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "c:\python38\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "c:\python38\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\python38\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the specified file。
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at D:\app\ros2\Lib\site-packages\launch\launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "D:\app\ros2\Lib\site-packages\launch\launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "D:\app\ros2\Lib\site-packages\launch\utilities\visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "D:\app\ros2\Lib\site-packages\launch\action.py", line 108, in visit
    return self.execute(context)
  File "D:\app\ros2\Lib\site-packages\launch\actions\opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "D:\app\ros2\Lib\site-packages\launch\actions\execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher.EXE-1]: process started with pid [15704]
[robot_state_publisher.EXE-1] Parsing robot urdf xml string.
[robot_state_publisher.EXE-1] [INFO] [1608517542.170447000] [robot_state_publisher]: got segment
[ERROR] [robot_state_publisher.EXE-1]: process[robot_state_publisher.EXE-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher.EXE-1]: sending signal 'SIGTERM' to process[robot_state_publisher.EXE-1]
[ERROR] [robot_state_publisher.EXE-1]: process has died [pid 15704, exit code 1, cmd 'D:\app\ros2\lib\robot_state_publisher\robot_state_publisher.EXE --ros-args --params-file C:\Users\ADMINI~1\AppData\Local\Temp\launch_params__5j98y1k'].

dzywater, 2020-12-21:
whatever launch scripts failed if import following classes:

from webots_ros2_core.trajectory_follower import TrajectoryFollower
from webots_ros2_core.joint_state_publisher import JointStatePublisher
from webots_ros2_core.webots_node import WebotsNode  #InheritFrom Node
from webots_ros2_core.webots_robotic_arm_node import WebotsRoboticArmNode #InheritFrom WebotsNode
from webots_ros2_core.webots_differential_drive_node import WebotsDifferentialDriveNode  #InheritFrom WebotsNode

So, I think these classes have something not compatible with webots2021a

dzywater, 2020-12-21:
I use the latest webots_ros2, then it's ok with webots2021a

lukicdarkoo, 2020-12-21:
`webots_ros2` for Webots R2021a is available only in the `master` branch, but it will be realesed in the next Foxy sync