cyberbotics/webots_ros2 #182: Controllers spawned through ros2 not working for Webots: in Windows 10

prabinrath, 2021-01-02:
I installed Webots and ros2 on my Win10 system. I tried to install the webots_ros2 pkg on my local workspace and launched the demo file 'ros2 launch webots_ros2_universal_robot universal_robot_multiple.launch.py' for UR robot simulation. Webots opens and the conveyor belt starts moving but the extern controllers for ur robots do not work and hence the robots don't move. 

When I press CTRL+C in PowerShell I get the following output

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

Loading personal and system profiles took 1135ms.
PS C:\Users\Admin> ros2 launch webots_ros2_universal_robot universal_robot_multiple.launch.py
[INFO] [launch]: All log files can be found below C:\Users\Admin\.ros/log\2021-01-02-19-28-48-433426-DESKTOP-Q4VN964-13284
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [universal_robot_multiple.wbt --batch --mode=realtime-1]: process started with pid [7848]
[INFO] [webots_robotic_arm_node.EXE-2]: process started with pid [2640]
[INFO] [webots_robotic_arm_node.EXE-3]: process started with pid [8512]
[universal_robot_multiple.wbt --batch --mode=realtime-1] OpenGL error: GL_INVALID_ENUM
[universal_robot_multiple.wbt --batch --mode=realtime-1] OpenGL error: GL_INVALID_ENUM
[universal_robot_multiple.wbt --batch --mode=realtime-1] OpenGL error: GL_INVALID_ENUM
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [universal_robot_multiple.wbt --batch --mode=realtime-1]: process has died [pid 7848, exit code 3221225786, cmd 'C:\Users\Admin\AppData\Local\Programs\Webots\msys64\mingw64\bin\webots C:\Users\Admin\Documents\ros2_ws\install\share\webots_ros2_universal_robot\worlds\universal_robot_multiple.wbt --batch --mode=realtime'].
[universal_robot_multiple.wbt --batch --mode=realtime-1] ^C
[ERROR] [webots_robotic_arm_node.EXE-3]: process has died [pid 8512, exit code 1, cmd 'C:\Users\Admin\Documents\ros2_ws\install\lib\webots_ros2_core\webots_robotic_arm_node.EXE --webots-robot-name=UR5e --ros-args -r __ns:=/UR5e --params-file C:\Users\Admin\AppData\Local\Temp\launch_params_tzxm7hn3'].
[webots_robotic_arm_node.EXE-3] **Cannot open file: C:\Users\Admin\AppData\Local\Temp/webots-13688/WEBOTS_SERVER (retrying in 1 second)**

[ERROR] [webots_robotic_arm_node.EXE-2]: process has died [pid 2640, exit code 1, cmd 'C:\Users\Admin\Documents\ros2_ws\install\lib\webots_ros2_core\webots_robotic_arm_node.EXE --webots-robot-name=UR3e --ros-args -r __ns:=/UR3e --params-file C:\Users\Admin\AppData\Local\Temp\launch_params_8wummeqt'].
[webots_robotic_arm_node.EXE-2] **Cannot open file: C:\Users\Admin\AppData\Local\Temp/webots-13688/WEBOTS_SERVER (retrying in 1 second)**

PS C:\Users\Admin>
```

I am unable to understand why this **Cannot open file: C:\Users\Admin\AppData\Local\Temp/webots-13688/WEBOTS_SERVER (retrying in 1 second)** message shows up for each controller spawn up! I tried running from Powershell Admin but got similar output. 

For once (maybe by chance) it started working and I was able to see the robots working but then subsequent trials were not working.

PS: I had to modify `webots_ros2_core/webots_ros2_core/utils.py` file in order to get the launch file working in the first place. `webotsw.exe` executable path was not added to Path system variable so I had to do that manually first. Then made the following changes to run the webots application through ros2 launch.

1.  Added `from pathlib import Path` and replaced `os.environ['HOME']` in line 68 with `str(Path.home())`. `os.environ['HOME']` was giving a KeyError in python3.
2.  Replaced `subprocess.check_output(['which', 'webots'])` in line 59 with `subprocess.check_output(['where', 'webotsw'])` because which is available in Linux. Windows alternative was `where` command. The executable name in Windows was `webotsw.exe` and not `webots.exe`.

If the above changes might have been the cause for the issue I am facing please let me know. Any help will be highly appreciated :). Thanks 

**My System config**
 - Webots Version:  R2020a
 - ROS Version: Foxy
 - Operating System: Windows 10 (Both Webots and ROS2 are installed in Windows 10)
 - Graphics Card: AMD Radon M360 (pretty old one but it works)

lukicdarkoo, 2021-01-02:
It fails here:
```log
[ERROR] [universal_robot_multiple.wbt --batch --mode=realtime-1]: process has died [pid 7848, exit code 3221225786
```

Could you please share the output of the following command:
```log
C:\Users\Admin\AppData\Local\Programs\Webots\msys64\mingw64\bin\webots C:\Users\Admin\Documents\ros2_ws\install\share\webots_ros2_universal_robot\worlds\universal_robot_multiple.wbt --batch --mode=realtime
```

prabinrath, 2021-01-03:
Hi @lukicdarkoo Thanks for the quick reply. Here is the output of the command you suggested.

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

Loading personal and system profiles took 1834ms.
PS C:\Users\Admin> C:\Users\Admin\AppData\Local\Programs\Webots\msys64\mingw64\bin\webots C:\Users\Admin\Documents\ros2_ws\install\share\webots_ros2_universal_robot\worlds\universal_robot_multiple.wbt --batch --mode=realtime
OpenGL error: GL_INVALID_ENUM
OpenGL error: GL_INVALID_ENUM
OpenGL error: GL_INVALID_ENUM
```

This opens up the Webots application with the environment but the conveyer belt isn't moving.  Below is a screenshot of the Webots app window.

[image]

I investigated a bit further and found that the location **C:\Users\Admin\AppData\Local\Temp/webots-13688/** does exists. It gets created when I do ros2 launch. However, I don't know what is preventing it from accessing WEBOTS_SERVER  file.

prabinrath, 2021-01-03:
```
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [universal_robot_multiple.wbt --batch --mode=realtime-1]: process has died [pid 7848, exit code 3221225786, cmd 'C:\Users\Admin\AppData\Local\Programs\Webots\msys64\mingw64\bin\webots C:\Users\Admin\Documents\ros2_ws\install\share\webots_ros2_universal_robot\worlds\universal_robot_multiple.wbt --batch --mode=realtime'].
[universal_robot_multiple.wbt --batch --mode=realtime-1]
```

@lukicdarkoo This is expected because I pressed CTRL+C to exit. Only then it shows `[ERROR] [universal_robot_multiple.wbt --batch --mode=realtime-1]: process has died [pid 7848, exit code 3221225786` an not when I launch it. 

I am  worried about 
```
[ERROR] [webots_robotic_arm_node.EXE-3]: process has died [pid 8512, exit code 1, cmd 'C:\Users\Admin\Documents\ros2_ws\install\lib\webots_ros2_core\webots_robotic_arm_node.EXE --webots-robot-name=UR5e --ros-args -r __ns:=/UR5e --params-file C:\Users\Admin\AppData\Local\Temp\launch_params_tzxm7hn3'].
[webots_robotic_arm_node.EXE-3]  Cannot open file: C:\Users\Admin\AppData\Local\Temp/webots-13688/WEBOTS_SERVER (retrying in 1 second)

[ERROR] [webots_robotic_arm_node.EXE-2]: process has died [pid 2640, exit code 1, cmd 'C:\Users\Admin\Documents\ros2_ws\install\lib\webots_ros2_core\webots_robotic_arm_node.EXE --webots-robot-name=UR3e --ros-args -r __ns:=/UR3e --params-file C:\Users\Admin\AppData\Local\Temp\launch_params_8wummeqt'].
[webots_robotic_arm_node.EXE-2]  Cannot open file: C:\Users\Admin\AppData\Local\Temp/webots-13688/WEBOTS_SERVER (retrying in 1 second)
```

As I don't understand why the controller isn't able to open the specified file while it does exist in TEMP folder.

lukicdarkoo, 2021-01-04:
I have just tested it on my Windows machine and it works fine (from here #186). Keep in mind that the arms are not supposed to move, the controllers only create topics and actions necessary for control.

However, since the belt is not moving it means that controllers have never started. Therefore, I am not sure what is causing your problem. The only difference is that I am using [ROS on Windows](https://discourse.ros.org/t/ros-on-windows-foxy-release-v20201211-0-0-2012290002/18241).

- Could you confirm that the sample Python controller (Open Webots, `File > Open Sample World...` and search for `python` example). The internal controller also uses `WEBOTS_SERVER` under-the-hood.
- Could you try [ROS on Windows](https://discourse.ros.org/t/ros-on-windows-foxy-release-v20201211-0-0-2012290002/18241).

PR #186 (2021-01-05): Various refactors and improvements
This PR has to do the following:
- [x] Make the package work on Windows (fixes #182) 
  - [x] Get rid of `tkinter`, so the package can be used comfortably with [ROS on Windows](https://github.com/ms-iot/ROSOnWindows).
  - [x] Add progress bar for downloading 
  - [x] Fix the home path on Windows 
  - [x] Replace `which` with `where` on Windows 
- [x] Introduce the notion of minimum and target Webots version (fixes #185).
- [x] Deprecate `TfPublisher` in favor of our new transform publisher based on `get_urdf()`. 
- [x] Enforce the camel-case project-wide (and [... cut]
closed 2021-01-05 by lukicdarkoo