cyberbotics/webots_ros2 #183: Build failed, missing 'lark'

Simon-Steinmann, 2021-01-02:
**Describe the Bug**
I followed the build instructions here https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install and got the following error:

Of Note:
I'm using ros2-foxy and a conda virtual environment.

```bash
(ros2) simon@simon-Legion-Y540:~/ws_ros2$ rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO
#All required rosdeps installed successfully
(ros2) simon@simon-Legion-Y540:~/ws_ros2$ colcon build
Starting >>> webots_ros2_msgs
Starting >>> webots_ros2_ur_e_description
Starting >>> webots_ros2_importer
Finished <<< webots_ros2_ur_e_description [0.77s]                     
Finished <<< webots_ros2_importer [0.77s]
--- stderr: webots_ros2_msgs                          
CMake Warning at /opt/ros/foxy/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:235 (add_library):
  Cannot generate a safe runtime search path for target
  webots_ros2_msgs__rosidl_typesupport_fastrtps_c__pyext because files in
  some directories may conflict with libraries in implicit directories:

    runtime library [libssl.so.1.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/simon/anaconda3/envs/ros2/lib
    runtime library [libcrypto.so.1.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/simon/anaconda3/envs/ros2/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:34 (rosidl_generate_interfaces)

Traceback (most recent call last):
  File "/opt/ros/foxy/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 8, in <module>
    from rosidl_generator_c import generate_c
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_c/__init__.py", line 15, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
Traceback (most recent call last):
  File "/opt/ros/foxy/share/rosidl_generator_cpp/cmake/../../../lib/rosidl_generator_cpp/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import generate_files
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
make[2]: *** [CMakeFiles/webots_ros2_msgs__rosidl_generator_c.dir/build.make:98: rosidl_generator_c/webots_ros2_msgs/srv/set_differential_wheel_speed.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:334: CMakeFiles/webots_ros2_msgs__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/webots_ros2_msgs__cpp.dir/build.make:106: rosidl_generator_cpp/webots_ros2_msgs/srv/set_differential_wheel_speed.hpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:308: CMakeFiles/webots_ros2_msgs__cpp.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
---
Failed   <<< webots_ros2_msgs [1.10s, exited with code 2]
                                 
Summary: 2 packages finished [1.25s]
  1 package failed: webots_ros2_msgs
  1 package had stderr output: webots_ros2_msgs
  10 packages not processed

```

It was a simple fix by installing it with
`pip3 install lark`

After his, everything compiled successfully.

closed 2021-01-02 by lukicdarkoo

ManuelZ, 2021-05-24:
I just followed the instructions and got the same error. 

```
Starting >>> webots_ros2_msgs
Starting >>> webots_ros2_ur_e_description
Starting >>> webots_ros2_importer
Finished <<< webots_ros2_importer [0.90s]                                                                                             
Finished <<< webots_ros2_ur_e_description [0.95s]
--- stderr: webots_ros2_msgs                             
Traceback (most recent call last):
  File "/opt/ros/foxy/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c", line 8, in <module>
    from rosidl_generator_c import generate_c
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_c/__init__.py", line 15, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
Traceback (most recent call last):
  File "/opt/ros/foxy/share/rosidl_generator_cpp/cmake/../../../lib/rosidl_generator_cpp/rosidl_generator_cpp", line 8, in <module>
    from rosidl_generator_cpp import generate_cpp
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_cpp/__init__.py", line 17, in <module>
    from rosidl_cmake import generate_files
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
make[2]: *** [CMakeFiles/webots_ros2_msgs__cpp.dir/build.make:194: rosidl_generator_cpp/webots_ros2_msgs/msg/wb_camera_recognition_object.hpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:361: CMakeFiles/webots_ros2_msgs__cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/webots_ros2_msgs__rosidl_generator_c.dir/build.make:178: rosidl_generator_c/webots_ros2_msgs/msg/wb_camera_recognition_object.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:387: CMakeFiles/webots_ros2_msgs__rosidl_generator_c.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webots_ros2_msgs [3.16s, exited with code 2]

Summary: 2 packages finished [3.61s]
  1 package failed: webots_ros2_msgs
  1 package had stderr output: webots_ros2_msgs
  11 packages not processed

```

lukicdarkoo, 2021-05-25:
@ManuelZ This is a bit different, it seems that dependencies of the `rosidl_parser` package are correctly installed. Did you run the following?
```bash
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

PR #184 (2021-01-02): Add `lark` dependency
Fixes #183

`lark` is dependency of `launch` that typically gets installed with `ros-base`, so we didn't notice it earlier:
https://github.com/ros2/launch/blob/55ca523ff0053e1c7932debaa7366e33180c23a6/launch/package.xml#L17