cyberbotics/webots_ros2 #185: Webots_ros2 requires specific Webots version

Simon-Steinmann, 2021-01-03:
**Describe the Bug**
I found, that webots_ros2 only works with the exact version of Webots specified here:
https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_core/webots_ros2_core/utils.py#L35

As of yesterday, the master branch of webots_ros2 required 2020b,
I built the develop branch, so I could use 2021a, However this did not work, as I had compiled the latest develop branch of Webots "2021b"
So I recompiled the master branch of Webots. It still did not work, as webots_ros2 requires "2021a" and the I now had "2021 Revision 1"

**Expected behavior**
- building both master branches together should work
- webots_ros2 should accept newer versions (perhaps with a warning)

**Proposed solutions**
- Turn the hard rquirement into a soft requirement, allowing the user to use any version they want and simply warn them if not the expected version was found. The popup window should probably include the Version that was actually found
- Add multiple valid versions to cover develop and built versions. This could look like this:

```python
def get_required_webots_version():
    """Return the Webots version compatible with this version of the package."""
    return ['R2021a', 'R2021a revision 1', 'R2021b']
```

lukicdarkoo, 2021-01-03:
The current situation is:
- the `master` branch of `webots_ros2` targets Webots on the `master` branch
- the `develop` branch of `webots_ros2` targets Webots on the `develop` branch
- the upstream `webots_ros2` package will target the latest stable Webots as soon as a new Foxy/Rolling sync is out

This way it is easier for us to maintain the `webots_ros2` package as we don't have to worry about backward/forward compatibility. I am afraid to create a mess by allowing users to use any version of Webots. For example, the current `webots_ros2` package (`develop` and `master`) cannot work with Webots R2020b or earlier as the `getDevice()` method doesn't exist in Webots R2020b or earlier.

Therefore, we should carefully design this feature and thoroughly consider whether it is really needed.

Simon-Steinmann, 2021-01-04:
I understand that webots_ros2 versions are not backwards compatible, however, shouldn't they be forward compatible? For example, webots_ros2 that works with 2020b shoudl work with 2021a as well right? And especially there should be no difference between installing and building the master branch ("2021a" vs "2021a revision 1")

lukicdarkoo, 2021-01-04:
That sounds good, we can define the recommended and minimum version.

stefaniapedrazzi, 2021-01-04:
Just note that is not guaranteed that the Webots R2020b API is forward compatible with R2021a API so I would also expect that `webots_ros2` that works with R2020b could not work with R2021a.

But between revisions ("2021a" and "2021a revision 1") the API compatibility is ensured.

PR #186 (2021-01-05): Various refactors and improvements
This PR has to do the following:
- [x] Make the package work on Windows (fixes #182) 
  - [x] Get rid of `tkinter`, so the package can be used comfortably with [ROS on Windows](https://github.com/ms-iot/ROSOnWindows).
  - [x] Add progress bar for downloading 
  - [x] Fix the home path on Windows 
  - [x] Replace `which` with `where` on Windows 
- [x] Introduce the notion of minimum and target Webots version (fixes #185).
- [x] Deprecate `TfPublisher` in favor of our new transform publisher based on `get_urdf()`. 
- [x] Enforce the camel-case project-wide (and [... cut]
closed 2021-01-05 by lukicdarkoo