cyberbotics/webots_ros2 #188: webots_ros2_epuck - windows 10 ros2 foxy - fail to find world files

anastasiapan, 2021-01-05:
**Describe the Bug**
Webots ros2 (foxy) builds fine with colcon on Windows 10. After sourcing, I try to run the tutorial (ros2 launch webots_ros2_epuck robot_launch.py) and it fails to find a file - I believe it is the world file.
Error message: [ERROR] [launch]: Caught exception in launch (see debug for traceback): [WinError 2] The system cannot find the file specified. 
The world exists and opens through the webots interface but ros cannot launch it. I believe that it is the world file because I print out both package_dir and world and then I get the error after a LaunchConfiguration is instantiated.
package_dir: C:\dev\webots_ws\install\webots_ros2_epuck\share\webots_ros2_epuck

**Steps to Reproduce**
1. I followed these instructions:
[https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#windows](url)
I did not run rosdep update according to [https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/](url)
2. colcon build - runs ok
3. call install\local_setup.bat
4. ros2 launch webots_ros2_epuck robot_launch.py

**Expected behavior**
As described here: [https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-E-puck-for-ROS2-Beginners#simulated-robot](url)

**Affected Packages**
List of affected packages:
  - webots_ros2_epuck (this is the only one I tested)

**System**
 - Webots Version: [R2021a]
 - ROS Version: [e.g., Foxy] 
 - Operating System: [Windows 10]
 - Graphics Card: [Nvidia Geforce RTX 2060]

lukicdarkoo, 2021-01-05:
Could try `webots_ros2` from #186 (branch `refactor-compability`)? There, I am finalizing support for Widows.

Since it is subject to change, please use d6d4e4a

anastasiapan, 2021-01-05:
I used the refactor-compabilty branch (commit d6d4e4a) and it looks like that indeed the world launcher fails. These are my error messages:

[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: nul
[INFO] [robot_state_publisher.EXE-1]: process started with pid [6584]
[INFO] [epuck_world.wbt --batch --mode=realtime-2]: process started with pid [18228]
[INFO] [driver.EXE-3]: process started with pid [1452]
[ERROR] [epuck_world.wbt --batch --mode=realtime-2]: process has died [pid 18228, exit code 1, cmd 'C:\Program Files\Webots\msys64\mingw64\bin\webots C:\dev\webots_ws_fix\src\webots_ros2\install\webots_ros2_epuck\share\webots_ros2_epuck\worlds\epuck_world.wbt --batch --mode=realtime'].
[epuck_world.wbt --batch --mode=realtime-2] 'C:\Program' is not recognized as an internal or external command,
[epuck_world.wbt --batch --mode=realtime-2] operable program or batch file.

lukicdarkoo, 2021-01-05:
I think it is because of the spaces in your path. Let me fix that.

lukicdarkoo, 2021-01-05:
Please check with 5305b14

anastasiapan, 2021-01-05:
So it turns out that it is the spacing in 'Program Files' directory. I tried the proposed fixes none of them worked for me. I found an easy (but not so smart) fix I just installed webots in a different directory than Program Files (with no spaces) and changed WEBOTS_HOME environment variable. I will close the issue now. Thanks for the indication.

closed 2021-01-05 by anastasiapan

lukicdarkoo, 2021-01-05:
@anastasiapan I am glad you found a workaround, but I would like to fix it. I tried to run on my machine with Webots installed at `C:\Program Files\Webots\msys64\mingw64\bin` and it worked fine. It's strange that the fix doesn't solve the issue for you as well.