cyberbotics/webots_ros2 #196: ROS2 foxy on Docker container and webots on Windows 10

anastasiapan, 2021-01-20:
**Describe the Bug**
Currently I have ROS2 foxy and the teleop-twist-keyboard package on a docker container and what I want to do is to publish the keys from the container and move the robot on webots which is installed on Windows 10. I have verified that teleop-twist works fine with the webots_ros2_epuck package if I run them both locally on Windows. But I wish to run the keyboard joystick on the container. I also tried to echo the topic and I can see that it is published normally when I run it locally but not when I run it on the container.

**Steps to Reproduce**
1. Build and start a docker container with ros-foxy:base (https://hub.docker.com/_/Ros)
2. Launch webots epuck example:
`ros2 launch webots_ros2_epuck robot_launch.py`
3. Start the teleop-twist-keyborad with ros2 run
4. Nothing happens on the webots

**Expected behavior**
To move epuck with the keyborad as you can do when everything is run locally.

**Affected Packages**
List of affected packages:
  - webots_ros2_epuck

**System**
 - Webots Version: [R2021a]
 - ROS Version: [Foxy] 
 - Operating System: [Windows 10]
 - Graphics Card: [Nvidia Geforce RTX 2060]

**Additional context**
Some things that I tried:

One thing I tried but didn't fix is to change the RMW_IMPLEMENTATION to eProsima (https://index.ros.org/doc/ros2/Concepts/DDS-and-ROS-middleware-implementations/). It should be the default value though. Having said that I am not at all experienced with DDS.

set RMW_IMPLEMENTATION=rmw_fastrtps_cpp

I have also tried to run the docker image with host network as below:
docker run -it --name some_name --network="host" container

lukicdarkoo, 2021-01-20:
This doesn't look related to Webots. To confirm, you can run:
```
ros2 topic pub /test std_msgs/String 'data: 1'
```
in your Docker and the following command in your Windows machine:
```
ros2 topic echo /test
```

If you don't receive anything in the Windows machine it means that is Docker/Windows/ROS2 related issue.

This may help:
https://stackoverflow.com/a/42425642/1983050

If not, I advise you to post the question on:
https://answers.ros.org/questions/

anastasiapan, 2021-01-20:
Thank you for your answer! And you are indeed right it is not only with Webots it is with everything I try to publish and subscribe with ROS so I will check there too.

closed 2021-01-20 by lukicdarkoo