cyberbotics/webots_ros2 #197: Webots epuck driver crashes when launching ros2 with Cyclone DDS

anastasiapan, 2021-02-01:
**Describe the Bug**
Architecture: I publish a message (/cmd_vel) from a Docker container to Windows host. The messages go through with some configuration I did using Cyclone DDS. The problem is that when launching webots like this:
`ros2 launch webots_ros2_epuck robot_launch.py` epuck example crashes and doesn't get the cmd_vel message. But if I run webots_ros2 from Windows without using my configuration (which I provide below) the messages go through and epuck moves. Instead I get this error:
`[ERROR] [driver.EXE-3]: process has died [pid 15168, exit code 1, cmd 'C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file nul --params-file C:\Users\ANASTA~1\AppData\Local\Temp\launch_params_53shnn8b'].
[driver.EXE-3] 1612164240.708307 [0]       3640: config: Domain/General/NetworkInterfaceAddress/#text: 10.1.1.74 {0}`

I tried to run the cmd that gives me the error alone:
`C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file nul --params-file C:\Users\ANASTA~1\AppData\Local\Temp\launch_params_53shnn8b`

 and I got this:
`usage: driver [-h] [--webots-robot-name WEBOTS_ROBOT_NAME]
driver: error: argument --webots-robot-name: expected one argument`

**Steps to Reproduce**
1. Build Docker image:
```
FROM ros:foxy

# install ros package
RUN apt-get update && apt-get install -y \
      ros-foxy-rmw-cyclonedds-cpp \
      ros-foxy-demo-nodes-cpp \
      ros-foxy-demo-nodes-py && \
    rm -rf /var/lib/apt/lists/*

ENV RMW_IMPLEMENTATION="rmw_cyclonedds_cpp"
ENV HOST_ADDR="10.1.1.74"

# Using participant index
ENV CYCLONEDDS_URI="<CycloneDDS><Domain id='any'><General><ExternalNetworkAddress>${HOST_ADDR}</ExternalNetworkAddress><AllowMulticast>false</AllowMulticast></General><Discovery><ParticipantIndex>1</ParticipantIndex><Peers><Peer address='${HOST_ADDR}'/></Peers></Discovery><Tracing><Verbosity>config</Verbosity><Out>stderr</Out></Tracing></Domain></CycloneDDS>"

```
2. Source ROS in the container
3. Open cmd terminal and source ROS and webots_ros2
4. Configure network with the following xml configuration:
(10.1.1.74 is my ethernet adapter address)

```xml
set "RMW_IMPLEMENTATION=rmw_cyclonedds_cpp"
set "HOST_ADDR=10.1.1.74"

REM Using participant index
set "CYCLONEDDS_URI=<CycloneDDS><Domain id='any'><General><NetworkInterfaceAddress>${HOST_ADDR}</NetworkInterfaceAddress><AllowMulticast>false</AllowMulticast></General><Discovery><ParticipantIndex>0</ParticipantIndex><Peers><Peer address='${HOST_ADDR}:7412'/></Peers></Discovery><Tracing><Verbosity>config</Verbosity><Out>stderr</Out></Tracing></Domain></CycloneDDS>"
```
5. Publish a velocity from the container
`ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}"`

Note: Without running Webots I can echo the cmd_vel topic from Windows side fine.

**Expected behavior**
Epuck gets the cmd_vel message and moves forward.

**Affected Packages**
List of affected packages:
  -  webots_ros2_epuck

**System**
 - Webots Version: [R2021a]
 - ROS Version: [foxy] 
 - Operating System: [Windows 10]
 - Graphics Card: [Nvidia Geforce RTX 2060]

lukicdarkoo, 2021-02-01:
What happens if you run:
```bash
C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul --params-file
```
(without `--webots-robot-name`)

anastasiapan, 2021-02-01:
Ok so If I run it exactly as yours I get a parsing error:
`C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul --params-file`
`[ERROR] [1612170111.398121200] [rcl]: Failed to parse global arguments`

If I add the file that is used above I get this:
`C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul --params-file C:\Users\ANASTA~1\AppData\Local\Temp\launch_params_53shnn8b`
`Cannot open pipe file: \\.\pipe\webots_13960_1030625947`
`Cannot connect to Webots on pipe: \\.\pipe\webots_13960_1030625947`

And this happens regardless if I activate my host environment configuration or not.

lukicdarkoo, 2021-02-01:
> If I add the file that is used above I get this:
> `C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul [...]

The pipe problem is related to this issue: https://github.com/cyberbotics/webots/issues/2641

Can you try to run a sample world in Webots (independent from ROS) and then again:
```bash
C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul --params-file
```

anastasiapan, 2021-02-01:
Ok so I am not sure if I understood correctly what do you mean independent from ROS but his is what I did:
I open the epuck tutorials world from Webots gui (WEBOTS_HOME -> projects -> samples -> tutorials -> worlds -> my_first_simulation.wbt
Then on a terminal I source ros and webots installation and activating my host environment:
`call C:\dev\ros2_foxy\ros2-windows\local_setup.bat`
`call C:\dev\webots_ws\install\local_setup.bat`
`call C:\dev\ros2_foxy\ros2-windows\host_env.bat`

then:
`C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-node-name webots_driver --ros-args --params-file nul --params-file  C:\Users\ANASTA~1\AppData\Local\Temp\launch_params_53shnn8b`

and I get my outputs from the xml file (because of the verbosity config) and no ouput related to webots. I does not crash though.

anastasiapan, 2021-02-03:
@lukicdarkoo Ok so I have not found a solution yet but I did find a workaround. I simply open Webots as you suggested an then run:
`C:\dev\webots_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-robot-name e-puck --webots-node-name webots_driver --ros-args --params-file nul --params-file C:\Users\ANASTA~1\AppData\Local\Temp\launch_params_53shnn8b` 

and then it gets command from my ros Docker container.

lukicdarkoo, 2021-02-03:
Thank you for letting us know. Hower, that behavior is odd. Using the launch file and running Webots + the driver should have the same outcome. Because the launch file is doing exactly that, running the Webots and the driver.

It happens only with CycloneDDS or with FastDDS as well?

anastasiapan, 2021-02-03:
@lukicdarkoo It is my DDS configuration that confuses it because if I do not set it, it works both with Cyclone and Fast. I understand that it is a problem very specific to my setup but I just posted the workaround in case there is someone working with containers like that.

mrbhjv, 2021-02-05:
Hi, I've tried this and found out no problems when running the [epuck example](https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-E-puck-for-ROS2-Beginners) using Cyclone.

However, when using FastDDS, I get the following output
```
[driver.EXE-3] Traceback (most recent call last):
[driver.EXE-3]   File "C:\Users\Bart\Documents\development\source\webots_ros2_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver-script.py", line 33, in <module>
[driver.EXE-3]     sys.exit(load_entry_point('webots-ros2-epuck==1.0.5', 'console_scripts', 'driver')())
[driver.EXE-3]   File "C:\Users\Bart\Documents\development\source\webots_ros2_ws\install\webots_ros2_epuck\Lib\site-packages\webots_ros2_epuck\driver.py", line 159, in main
[driver.EXE-3]     rclpy.init(args=args)
[driver.EXE-3]   File "C:\Users\Bart\Documents\development\ros2-windows\Lib\site-packages\rclpy\__init__.py", line 71, in init
[driver.EXE-3]     return context.init(args)
[driver.EXE-3]   File "C:\Users\Bart\Documents\development\ros2-windows\Lib\site-packages\rclpy\context.py", line 59, in init
[driver.EXE-3]     rclpy_implementation.rclpy_init(args if args is not None else sys.argv, capsule)
[driver.EXE-3] RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation: C:\Users\Bart\Documents\development\ros2-windows\bin/rmw_fastrtps_cpp.dll Exception: LoadLibrary error: 126, at C:\ci\ws\src\ros2\rcutils\src\shared_library.c:88, at C:\ci\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:64, at C:\ci\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:61
[ERROR] [driver.EXE-3]: process has died [pid 2744, exit code 1, cmd 'C:\Users\Bart\Documents\development\source\webots_ros2_ws\install\webots_ros2_epuck\lib\webots_ros2_epuck\driver.EXE --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file C:\Users\Bart\AppData\Local\Temp\launch_params_aqj4fl4g'].
[ERROR] [robot_state_publisher.EXE-1]: process has died [pid 2716, exit code 3221226505, cmd 'C:\Users\Bart\Documents\development\ros2-windows\lib\robot_state_publisher\robot_state_publisher.EXE --ros-args --params-file C:\Users\Bart\AppData\Local\Temp\launch_params_kzl5smx0'].
[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [epuck_world.wbt" --batch --mode=realtime-2]: process has died [pid 15144, exit code 3221225786, cmd '"C:\Users\Bart\Documents\development\Webots\msys64\mingw64\bin\webots" "C:\Users\Bart\Documents\development\source\webots_ros2_ws\install\webots_ros2_epuck\share\webots_ros2_epuck\worlds\epuck_world.wbt" --batch --mode=realtime']
```

Maybe this is related, as the error seems to be quite similar.

lukicdarkoo, 2021-02-05:
> RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation: C:\Users\Bart\Documents\development\ros2-windows\bin/rmw_fastrtps_cpp.dll...

This doesn't look related to Webots. However, if the simple `ros2 topic pub`/`ros2 topic echo` commands work, then we may be making some mess around here:
https://github.com/cyberbotics/webots_ros2/blob/8230dd586462c4754f5eb4fd2022f8428ff94a90/webots_ros2_core/webots_ros2_core/utils.py#L187

For me, the example works with FastDDS (I use ROS on Windows). Maybe you are falling into some edge case that I am not aware of.

anastasiapan, 2021-02-09:
Ok so after some more debugging I figured out that the robot state publisher doesn't launch. The issue seems to be that when I run with my specific DDS config I cannot run multiple nodes with ROS. So the first node launches (which is webots gui) and the next one crushes. And this is also the reason that when I run the driver script it works - because it starts only the driver node.

lukicdarkoo, 2021-02-09:
> with my specific DDS config I cannot run multiple nodes with ROS

Do you have any why?

anastasiapan, 2021-02-10:
> Do you have any why?

I think I do. To make my setup work I need to disable multicast (https://github.com/eclipse-cyclonedds/cyclonedds/issues/677) and expose some ports. In this way I believe that when I start ROS, the first node launched uses the participant index set for the specific DDS service instance, and the rest of them cannot go through since I am not using multicast. And that is the reason that when I start only the driver there are no problems - it is only one node. But I plan to work with the ros2 navigation stack on Webots and I need other nodes available too (e.g. the robot state publisher). Something I thought to do is set a new participant index for each node to be discoverable but this is not the most efficient/practical way to work with.

lukicdarkoo, 2021-02-10:
https://github.com/eclipse-cyclonedds/cyclonedds/issues/687
I believe this issue is related to the multi-node problem.

Thank you very much for the clarification. I will close this issue as it is not specific to Webots, but if you discover any problem related to Webots feel free to reopen it.

closed 2021-02-10 by lukicdarkoo

anastasiapan, 2021-02-10:
@lukicdarkoo I opened that issue since it is related to DDS. I know that it is not Webots related by now, but I just keep updating here too to inform about what breaks the setup. I will also link the solution in a new comment when I figure it out in case someone wants to work with containers.