cyberbotics/webots_ros2 #202: "ros2 launch webots_ros2_epuck robot_tools_launch.py rviz:=true nav:=true" does not work

qingtian1771, 2021-02-19:
I'm using ros2 foxy and webot 2021a to learn ros2 integration according this tutorial, I found following command can not be executed right.
ros2 launch webots_ros2_epuck robot_tools_launch.py rviz:=true nav:=true

rviz is displayed but I think is not right, it give me this warning:
```
Fixed Frame
No tf data. Actual error: Frame [map] does not exist.
```
**And webots did not start at all!**

And there is some info on the console output

`[controller_server-5] [INFO] [1613706007.018927746] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
`

**Screenshots**
[image]

**System**
 - Webots Version: [R2021a ]
 - ROS Version: [foxy] 
 - Operating System: [ubuntu 20.04]
 - Graphics Card: []

lukicdarkoo, 2021-02-19:
You have to use the `robot_with_tools_launch.py` file:
https://github.com/cyberbotics/webots_ros2/wiki/Tutorial-E-puck-for-ROS2-Beginners#simulated-robot

The `robot_tools_launch.py` file is suitable for the physical robot.

qingtian1771, 2021-02-19:
Thanks a lot, it works!

closed 2021-02-19 by qingtian1771