cyberbotics/webots_ros2 #209: add args for "--stream" for webots simulation streaming (temp workaround in comments)

WyattAutomation, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
**A feature for handling the "--stream" and other webots args for serving/accessing webots simulations from a browser via websocket and http:// need to be added**.  I could not find docs on this feature, nor where there is existing configurable functionality to add it to the ros2 launch command.  The webots docs for these args are here:  https://cyberbotics.com/doc/guide/starting-webots#streaming-server-options

**Describe the solution you'd like**
I'm open to adding to the existing args parsing/handling for the launcher myself and submitting a pr, if it's not already done or being addressed in another issue.

**Describe alternatives you've considered**
I've shoehorned-in a "--stream" argument to the string put together by webots_launcher,py for the webots command, and it streaming works fine.

**Additional context**
**For anyone wanting a quick bandaid to get simulation streaming working**
Change line 61 in webots_launcher.py (https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_core/webots_ros2_core/webots_launcher.py#L61)
from:
```
        command.append('--mode=' + context.perform_substitution(self.__mode))
```
to:
'''
        command.append('--stream="mode=mjpeg" --mode=' + context.perform_substitution(self.__mode))
'''
Then cd to your colcon workspace (contains your src folder) and run a full build from your workspace home:
```
colcon build
```
Then run your launchfile for the desired ros2 webots package, and follow the normal steps for accessing a webots streaming server per their docs https://cyberbotics.com/doc/guide/web-streaming#how-to-stream-a-webots-simulation

^that is not an ideal fix, just some breadcrumbs for anyone wanting this feature prior to a pr.

lukicdarkoo, 2021-03-01:
> Requesting a feature for handling the "--stream" and other webots args for serving/accessing webots simulations from a browser via websocket and http:// need to be added. I could not find docs on this feature, nor where there is existing configurable functionality to add it to the ros2 launch command

Unfortunately, It doesn't exist in the `webots_ros2` interface.

> I'm open to adding to the existing args parsing/handling for the launcher myself and submitting a pr, if it's not already done or being addressed in another issue.

Please go ahead:
- The argument should be added here: https://github.com/cyberbotics/webots_ros2/blob/d0f772b358252cda6b9e56768e5ec5cf7454a5ff/webots_ros2_core/webots_ros2_core/webots_launcher.py#L66
- And a launch argument here:
https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_core/launch/robot_launch.py

> Also note: **The Reset Simulation button in the GUI from webot's streaming viewer html, reproduces the same problem from #40 with extern controllers not being relaunched when simulation is relaunched**

I am not sure whether it is a good approach to restart the simulation only (Webots and the controller). Let's say you start the simulation and it reaches a state A_{simulation}, and the other nodes (e.g. navigation and mapping) reach the corresponding state A_{control}. If you restart the simulation your simulation will be in state B_{simulation} even though the other nodes are still in the state A_{control}. Therefore, the nodes will completely confused. Please correct me if I am wrong.

I was thinking to emit an event when the simulation is restarted/reset, so a user can decide how to handle it. Something similar to what we are doing right now for exiting the simulation:
https://github.com/cyberbotics/webots_ros2/blob/d0f772b358252cda6b9e56768e5ec5cf7454a5ff/webots_ros2_core/launch/robot_launch.py#L142

WyattAutomation, 2021-03-04:
I will do my best to put in a quality PR for the "--stream" args ASAP, thank you for the tips on where to start.  Pretty straightforward, I just need to contribute some effort here.

> I am not sure whether it is a good approach to restart the simulation only (Webots and the controller). Let's say you start the simulation and it reaches a state A_{simulation}, and the other nodes [...]

After becoming more familiar with webots over the past couple of weeks, I think I agree with this.  I assume what you're saying is that you wouldn't want ROS2 to get way out of sync with the progression of the webots simulation because of a disjointed "restart" between the webots simulation, ROS2 nodes that make use of the webots interface, and ROS2 nodes that aren't making use of the webots interface?  Seems like it would require almost an entire process exit and complete restart of the launch (which is where I was headed in terms of a "workaround" for this)

In the lines you mentioned:
```
        RegisterEventHandler(
            event_handler=launch.event_handlers.OnProcessExit(
                target_action=webots,
                on_exit=[EmitEvent(event=launch.events.Shutdown())],
            )
```
I am assuming this is simply saying "when this ros2 launch process exits, gracefully shutdown webots too".  

Is it possible to call an additional event handler, to handle whatever specific event(s) from webots occur when the "reset" button is activated from the gui, to trigger both gracefully shutting down webots, but also to exit and then "relaunch" the ROS2 launchfile itself (relaunch all associated nodes)?  Forgive my ignorance if this the wrong approach, or not a built-in functionality for the launch.event_handler in ROS2 (I am still building familiarity for best practices for general ROS2 usage).  

I did a quick once-over to review the classes from launch.event_handlers and didn't see anything right away that stuck out as being able to have a launchfile "relaunch" itself upon something like webots gui events.  I feel like I'm probably overlooking something simple either with how to handle this, or that a full relaunch may not even be necessary (or desirable). 

Thanks again for your feedback and for such a detailed/quality response; I am still familiarizing myself with webots and ROS2 simulation best-practices in general, so my questions here are likely part "feature request" and part "I didn't know it could do that".  Let me know your thoughts, and I'll proceed with the effort needed to get some of the core "--stream" args functionality established in the meantime.

lukicdarkoo, 2021-03-04:
> Is it possible to call an additional event handler, to handle whatever specific event(s) from webots occur when the "reset" button is activated from the gui

No :/ To be honest I have no idea how to implement it. We should explore how to implement a custom event in ROS 2 launch and how to pass the event from Webots. On the Webots side, we probably can implement whatever is needed to pass the events to the ROS 2 launch.

You can create a proof of concept before proposing changes to Webots. Like, implement a simple application that replaces Webots and emits necessary events to the ROS 2 launch (to restart nodes). Once you have the example working feel free to propose changes to Webots, open an issue, or PR.

WyattAutomation, 2021-03-04:
Will do.  
In the meantime, I just found this recent (and possibly relevant) "respawn" options added to execute_process.py; looks like it was just merged a couple weeks ago (my version of foxy was out of date and didn't have this): 

https://github.com/ros2/launch/commit/77f0ae9f05a0e34626c4fb13484a2688ef9574af

Looks like additional parameters for the launch description that may allow for conditional respawning of nodes; might be getting warmer here.  Going to try to utilize this to test out a "respawn" of a node upon closing webots. 

Then I'll look into how webots gets a "closed" state from the restart button, and how that can be differentiated from simply closing the window.  Some days ago I am certain I stumbled across the API docs for what the actual gui buttons returned--It's in a sea of bookmarks so bare with me and I'll see if I can't get a proof of concept rolling here in the next few hours.

lukicdarkoo, 2021-03-04:
a) Maybe we can create an action, similar to `ExecuteProcess` that reads from the standard output and fires an event if a certain log message is detected. The log message can be passed from the controller (using the `--stdout` flag to pass the standard output from the controller to Webots), so there are no changes needed in the in Webots.

b) Even simpler. Consider that the simulation reset kills the controller. Also, consider that the `ControllerLauncher` inherits the `Node` inherits the `ExecuteProcess` which exposes the `on_exit` callback. That means that we have been already detecting when the simulation is reset/reloaded by getting the `on_exit` callback in the `ControllerLauncher`. Everything is there, we should just make example :)

So here:
https://github.com/cyberbotics/webots_ros2/blob/d0f772b358252cda6b9e56768e5ec5cf7454a5ff/webots_ros2_core/launch/robot_launch.py#L109
we should add an event fired on the `on_exit` callback. On this event, we should restart the `ControllerLauncher` and all other relevant nodes (if possible).

ygoumaz, 2023-02-20:
These issues were solved in https://github.com/cyberbotics/webots_ros2/pull/345, https://github.com/cyberbotics/webots_ros2/pull/502 and https://github.com/cyberbotics/webots_ros2/pull/625.

closed 2023-02-20 by ygoumaz