cyberbotics/webots_ros2 #211: Tesla foxy unexpected keyword argument 'controller_class'

AleBurzio11, 2021-03-08:
When I run the robot_launch.py file I encounter the following error: (I am on ROS2-foxy)

`ros2 launch webots_ros2_tesla robot_launch.py 
[INFO] [launch]: All log files can be found below /home/ale/.ros/log/2021-03-08-10-52-38-366481-ale-MSI-13665
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /dev/null
[INFO] [tesla_world.wbt" --batch --mode=realtime-1]: process started with pid [13668]
[INFO] [tesla_driver-2]: process started with pid [13671]
[INFO] [lane_follower-3]: process started with pid [13673]
[tesla_driver-2] Traceback (most recent call last):
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/webots_ros2_tesla/tesla_driver", line 11, in <module>
[tesla_driver-2]     load_entry_point('webots-ros2-tesla==1.0.5', 'console_scripts', 'tesla_driver')()
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/python3.8/site-packages/webots_ros2_tesla/tesla_driver.py", line 38, in main
[tesla_driver-2]     driver = TeslaDriver(args=args)
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/python3.8/site-packages/webots_ros2_tesla/tesla_driver.py", line 25, in __init__
[tesla_driver-2]     super().__init__('tesla_driver', args, controller_class=Driver)
[tesla_driver-2] TypeError: __init__() got an unexpected keyword argument 'controller_class'
[ERROR] [tesla_driver-2]: process has died [pid 13671, exit code 1, cmd '/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/webots_ros2_tesla/tesla_driver --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file /tmp/launch_params_0us_vl5v']`

I'm pretty sure it's a dumb thing to do but I tried removing the argument from the init call and got the following errors:
`ros2 launch webots_ros2_tesla robot_launch.py 
[INFO] [launch]: All log files can be found below /home/ale/.ros/log/2021-03-08-10-57-26-359306-ale-MSI-15175
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /dev/null
[INFO] [tesla_world.wbt" --batch --mode=realtime-1]: process started with pid [15178]
[INFO] [tesla_driver-2]: process started with pid [15181]
[INFO] [lane_follower-3]: process started with pid [15183]
[tesla_driver-2] Cannot open file: /tmp/webots-15180-fQtJFV/WEBOTS_SERVER (retrying in 1 second)
[tesla_driver-2] Cannot open file: /tmp/webots-15180-fQtJFV/WEBOTS_SERVER (retrying in 2 seconds)
[tesla_driver-2] Cannot open file: /tmp/webots-15180-fQtJFV/WEBOTS_SERVER (retrying in 3 seconds)
[tesla_driver-2] Traceback (most recent call last):
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/webots_ros2_tesla/tesla_driver", line 11, in <module>
[tesla_driver-2]     load_entry_point('webots-ros2-tesla==1.0.5', 'console_scripts', 'tesla_driver')()
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/python3.8/site-packages/webots_ros2_tesla/tesla_driver.py", line 39, in main
[tesla_driver-2]     rclpy.spin(driver)
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
[tesla_driver-2]     executor.spin_once()
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 711, in spin_once
[tesla_driver-2]     raise handler.exception()
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
[tesla_driver-2]     self._handler.send(None)
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 426, in handler
[tesla_driver-2]     await call_coroutine(entity, arg)
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 340, in _execute_timer
[tesla_driver-2]     await await_or_execute(tmr.callback)
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 118, in await_or_execute
[tesla_driver-2]     return callback(*args)
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/webots_node.py", line 99, in __timer_callback
[tesla_driver-2]     self.step(self.timestep)
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/webots_node.py", line 83, in step
[tesla_driver-2]     self.__device_manager.step()
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/devices/device_manager.py", line 81, in step
[tesla_driver-2]     device.step()
[tesla_driver-2]   File "/home/ale/potholes-detection/ros2_ws/install/webots_ros2_core/lib/python3.8/site-packages/webots_ros2_core/devices/camera_device.py", line 112, in step
[tesla_driver-2]     self._image_publisher.publish(msg)
[tesla_driver-2]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/publisher.py", line 70, in publish
[tesla_driver-2]     _rclpy.rclpy_publish(capsule, msg)
[tesla_driver-2] TypeError: expected a sequence in 'data'
[ERROR] [tesla_driver-2]: process has died [pid 15181, exit code 1, cmd '/home/ale/potholes-detection/ros2_ws/install/webots_ros2_tesla/lib/webots_ros2_tesla/tesla_driver --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file /tmp/launch_params_qpd42waj'].`

I installed all dependencies with rosdep but maybe there is something wrong with my setup?
K also had to manually install the ackerman_msgs for foxy

lukicdarkoo, 2021-03-08:
Could you please check whether you have the `webots_ros2` Debian package installed?
```bash
apt list --installed | grep webots
```

It can happen that the compiled package and the Debian package clash causing some weird errors.

AleBurzio11, 2021-03-08:
It seems to be correctly installed:

```
apt list --installed | grep webots

ros-foxy-webots-ros2-abb/focal,now 1.0.5-1focal.20210225.220727 amd64 [installed,automatic]
ros-foxy-webots-ros2-core/focal,now 1.0.5-1focal.20210225.220645 amd64 [installed,automatic]
ros-foxy-webots-ros2-demos/focal,now 1.0.5-1focal.20210226.004438 amd64 [installed,automatic]
ros-foxy-webots-ros2-epuck/focal,now 1.0.5-1focal.20210226.004308 amd64 [installed,automatic]
ros-foxy-webots-ros2-examples/focal,now 1.0.5-1focal.20210225.220721 amd64 [installed,automatic]
ros-foxy-webots-ros2-importer/focal,now 1.0.5-1focal.20210225.205633 amd64 [installed,automatic]
ros-foxy-webots-ros2-msgs/focal,now 1.0.5-1focal.20210225.205553 amd64 [installed,automatic]
ros-foxy-webots-ros2-tiago/focal,now 1.0.5-1focal.20210226.004306 amd64 [installed,automatic]
ros-foxy-webots-ros2-turtlebot/focal,now 1.0.5-1focal.20210225.220728 amd64 [installed,automatic]
ros-foxy-webots-ros2-tutorials/focal,now 1.0.5-1focal.20210225.220755 amd64 [installed,automatic]
ros-foxy-webots-ros2-universal-robot/focal,now 1.0.5-1focal.20210226.004308 amd64 [installed,automatic]
ros-foxy-webots-ros2-ur-e-description/focal,now 1.0.5-1focal.20210125.194205 amd64 [installed,automatic]
ros-foxy-webots-ros2/focal,now 1.0.5-1focal.20210226.004519 amd64 [installed]
webots/unknown,now 2021a amd64 [installed]
```

Although now that I take a closer look the Tesla package is not installed

lukicdarkoo, 2021-03-08:
You should delete all `ros-foxy-webots-*` pacakges. The compiled Tesla package is maybe trying to include the old `webots_ros2_core` package that cannot handle some new features we needed to introduce.

AleBurzio11, 2021-03-08:
I purged all apt packages but still get the "unexpected keyword argument"

lukicdarkoo, 2021-03-08:
The problem is in here I believe:
https://github.com/cyberbotics/webots_ros2/blob/d0f772b358252cda6b9e56768e5ec5cf7454a5ff/webots_ros2_core/webots_ros2_core/webots_node.py#L46

I believe that your Tesla package is trying to access the old `webots_ros2_core` package that doesn't have the `controller_class` argument.

I am not sure what is the best way to debug it, maybe:
- Open a new terminal that doesn't have sourced anything ROS-related.
- Make sure your `webots_ros2` package is checked out to the `master` branch.
- Run `source /opt/ros/foxy/local_setup.bash`.
- In your ROS 2 workspace (`/home/ale/potholes-detection/ros2_ws/`) delete the cache `rm -rf build install logs`. Then, build it with `colcon build`.

AleBurzio11, 2021-03-08:
The last solution worked and now everything seems to be fine, thanks for the help!

lukicdarkoo, 2021-03-08:
Great! Good luck with the simulations!

Feel free to join our Discord community in case of minor technical issues:
https://discordapp.com/invite/nTWbN9m

closed 2021-03-08 by lukicdarkoo