cyberbotics/webots_ros2 #218: New bloom release of webots_ros2 to ROS 2 Rolling

clalancette, 2021-04-06:
We've recently made some changes to ROS 2 Rolling that requires a new bloom release of this message package. This will fix the failing package build on Rolling that you can see in https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__webots_ros2_msgs__ubuntu_focal_amd64__binary/43/ . Note that we only require a bloom release; a new source release should not be necessary. To re-release this package the following steps should be performed:

$ sudo apt-get update
$ sudo apt-get install python3-bloom
# Ensure that the bloom-version you are using is 0.10.3 by running 'bloom-release --version'
$ bloom-release --track rolling --rosdistro rolling webots_ros2

Please let me know if you have any questions about this. Thank you.

lukicdarkoo, 2021-04-07:
Thanks for letting us know.

Simply running `bloom-release --track rolling --rosdistro rolling webots_ros2` fails. Adding a note here to fix it later:
```
Failed to resolve python3-pep8 on rhel:8 with: Error running generator: Failed to resolve rosdep key 'python3-pep8', aborting.
python3-pep8 is depended on by these packages: ['webots_ros2_importer']
```

cottsay, 2021-04-09:
https://github.com/cyberbotics/urdf2webots/pull/116

lukicdarkoo, 2021-04-12:
@cottsay Thank you, I replaced the package in the Rolling distribution.

I have just published a new Rolling version. Unfortunately, it failed again for RHEL-8, now because of the `python3-collada` package. Any chance for the RHEL-8 version of `python3-collada` to be added?

cottsay, 2021-04-12:
There's a fair chance, yes. It seems that package is typically sourced from EPEL, so I filed this issue to try to get it added: https://bugzilla.redhat.com/1948652

This probably won't happen very soon. Thanks for giving it a shot and enumerating the issues, though.

cottsay, 2021-05-04:
@lukicdarkoo, the EPEL maintainers built python3-collada and the package is now stable. I added a rule to the rosdep database: ros/rosdistro#29400

From what I can tell, the next issue is `python-imaging`. RHEL 8 has very few Python 2 packages, and this is not one of them.

Looking at the packages, it appears that `webots_ros2_importer` is taking a dependency on `imaging` and `numpy` from Python 2. Since the `setup.py` will be processed as Python 3 in ROS 2, I think that these are actually Python 3 dependencies, and the `package.xml` needs to be updated. I'm not entirely sure, though.