cyberbotics/webots_ros2 #223: Problem with URDF when launching robot_launch.py from webots_ros2_epuck package

aomezzine, 2021-04-20:
Hi,
when I launch this command : ros2 launch webots_ros2_epuck robot_launch.py 
I got this problem :
[robot_state_publisher-1] Initialize urdf model from file: --ros-args
[robot_state_publisher-1] Could not open file [--ros-args] for parsing.
[robot_state_publisher-1] Unable to initialize urdf::model from --ros-args

Can you help me please.
Thanks

lukicdarkoo, 2021-04-27:
Which ROS and Webots version do you use?

aomezzine, 2021-04-27:
ROS2 eloquent
webots R2020b revision 1

lukicdarkoo, 2021-04-27:
Do you use the `eloquent` branch?
https://github.com/cyberbotics/webots_ros2/tree/eloquent

aomezzine, 2021-04-27:
Yes, I verify it

aomezzine, 2021-04-28:
Even when I launch this : ros2 launch webots_ros2_core robot_launch.py \
    world:=$(ros2 pkg prefix webots_ros2_universal_robot --share)/worlds/universal_robot_rviz.wbt

I got the same problem :
[robot_state_publisher-1] Initialize urdf model from file: --ros-args
[robot_state_publisher-1] Could not open file [--ros-args] for parsing.
[robot_state_publisher-1] Unable to initialize urdf::model from --ros-args

lukicdarkoo, 2021-04-28:
It looks like the Eloquent API has changed. However, Eloquent is deprecated, so it is not interesting for us to invest time in it.

Is it possible for you to switch to Foxy?

lukicdarkoo, 2021-07-22:
I am closing this due to inactivity, but feel free to reopen if needed. I strongly suggest upgrading to webots_ros2 1.1.0 and Webots R2021b

closed 2021-07-22 by lukicdarkoo