cyberbotics/webots_ros2 #224: Problem about coordinate system

BruceXSK, 2021-04-20:
I use the DeviceManager to manage the devices on my robot, include a Velodyne VLP-16 lidar.
The default coordinate system of Webots is NUE but the ROS is ENU. So there is some trouble with the PointCloud data.
I revised the "coordinateSystem" of "WorldInfo" to "ENU". And revised the xyz order of PointCloud2 message in cloud data publish function of `webots_ros2_core/devices/lidar_device.py` as:
```Python3
msg.fields = [
    PointField(name='y', offset=0, datatype=PointField.FLOAT32, count=1),
    PointField(name='z', offset=4, datatype=PointField.FLOAT32, count=1),
    PointField(name='x', offset=8, datatype=PointField.FLOAT32, count=1)
]
```

This will solve the problem for now. But I still wonder if there is a general practice to solve this coordinate system transformation problem.
Thanks a lot

lukicdarkoo, 2021-04-27:
ENU is actually a coordinate system of the world and it is independent of FLU (Forward, Left, Up), a convention used for ROS robots.

Unfortunately, at the moment, we don't have a cleaner solution. However, we are slowly converting all PROTOs to FLU. Our plan is to have all PROTOs (including robots, objects, and devices) using FLU convention.

BruceXSK, 2021-04-28:
> ENU is actually a coordinate system of the world and it is independent of FLU (Forward, Left, Up), a convention used for ROS robots.
>
> Unfortunately, at the moment, we don't have a cleaner solution. [...]

I've got it. Thank you

closed 2021-04-28 by BruceXSK