cyberbotics/webots_ros2 #226: turtlebot_driver ERROR during 'Navigate TurtleBot3' tutorial

matteoderose, 2021-05-01:
**Describe the Bug**
I'm trying to run the simulation of TurtleBot3 navigation as in this link: [https://github.com/cyberbotics/webots_ros2/wiki/Navigate-TurtleBot3](url)
At the end Rviz does not get properly all the tf transforms. However, I think the problem is due to the webots launch.

**Steps to Reproduce**
Exactly the ones in the link above.

**Actual behavior**
On Terminal 1, where I run `ros2 launch webots_ros2_turtlebot robot_launch.py`, I get:
```
[INFO] [launch]: All log files can be found below /home/matteodr/.ros/log/2021-05-01-19-44-08-499865-matteodr-8881
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /dev/null
[INFO] [robot_state_publisher-1]: process started with pid [8884]
[INFO] [turtlebot3_burger_example.wbt" --batch --mode=realtime-2]: process started with pid [8886]
[INFO] [turtlebot_driver-3]: process started with pid [8889]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] [INFO] [1619891048.869360630] [robot_state_publisher]: got segment 
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 1 second)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 2 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 3 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 4 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 5 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 6 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 7 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 8 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 9 seconds)
[turtlebot_driver-3] Cannot open file: /tmp/webots-8888-MAHsmv/WEBOTS_SERVER (retrying in 10 seconds)
[turtlebot_driver-3] Impossible to communicate with Webots: aborting
[turtlebot_driver-3] Cannot connect to Webots: no pipe defined
[ERROR] [turtlebot_driver-3]: process has died [pid 8889, exit code 1, cmd '/home/matteodr/ros2_webots_ws/install/webots_ros2_turtlebot/lib/webots_ros2_turtlebot/turtlebot_driver --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file /tmp/launch_params_eb0xq060'].
```

Then in Terminal 2 I run
```
 export TURTLEBOT3_MODEL='burger'
ros2 launch turtlebot3_navigation2 navigation2.launch.py \
    use_sim_time:=true \
    map:=$(ros2 pkg prefix webots_ros2_turtlebot --share)/resource/turtlebot3_burger_example_map.yaml
```
and I get:

```
[INFO] [launch]: All log files can be found below /home/matteodr/.ros/log/2021-05-01-20-13-27-528583-matteodr-9811
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [9813]
[INFO] [amcl-2]: process started with pid [9815]
[INFO] [lifecycle_manager-3]: process started with pid [9817]
[INFO] [controller_server-4]: process started with pid [9819]
[INFO] [planner_server-5]: process started with pid [9821]
[INFO] [recoveries_server-6]: process started with pid [9828]
[INFO] [bt_navigator-7]: process started with pid [9834]
[INFO] [waypoint_follower-8]: process started with pid [9842]
[INFO] [lifecycle_manager-9]: process started with pid [9855]
[INFO] [rviz2-10]: process started with pid [9860]
[lifecycle_manager-3] [INFO] [1619892808.956600109] [lifecycle_manager_localization]: Creating
[lifecycle_manager-3] [INFO] [1619892809.049254318] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[lifecycle_manager-3] [INFO] [1619892809.097115830] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1619892809.099240430] [lifecycle_manager_localization]: Configuring map_server
[map_server-1] [INFO] [1619892809.305689019] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1619892809.312349193] [map_server]: Creating
[map_server-1] [INFO] [1619892809.313981686] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: /home/matteodr/ros2_webots_ws/install/webots_ros2_turtlebot/share/webots_ros2_turtlebot/resource/turtlebot3_burger_example_map.yaml
[map_server-1] [DEBUG] [map_io]: resolution: 0.05
[map_server-1] [DEBUG] [map_io]: origin[0]: -2.06
[map_server-1] [DEBUG] [map_io]: origin[1]: -6.84
[map_server-1] [DEBUG] [map_io]: origin[2]: 0
[... 167 lines cut ...]
[rviz2-10] [INFO] [1619892818.269738348] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[controller_server-4] [INFO] [1619892818.509278363] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-4] [INFO] [1619892819.008375650] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-10] [INFO] [1619892819.270188217] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-10] [INFO] [1619892819.270409641] [rviz2]: Sending lifecycle_manager_navigation/is_active request
[controller_server-4] [INFO] [1619892819.508412918] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[controller_server-4] [INFO] [1619892820.008533495] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
```

My attention focused especially on:
'[controller_server-4] [INFO] [1619892816.008224584] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist'
I think this is due to the fact that webots is not able to connect.

**Screenshots**
This is my Rviz scene tree:
[image]

**System**
 - Webots Version: R2021a - 15 Dec 2020
 - ROS version: ROS2 Foxy
 - Operating System: Linux Ubuntu 20.04 on Virtual Machine through VirtualBox
 - Graphics Card: Intel(R) HD Graphics 4000

**Additional context**
I think the problem might be solved by increasing the time for which turtlebot_driver waits for webots to be completely up, which is actually 10 seconds, but I don't know how to do it.

lukicdarkoo, 2021-05-02:
The problem seems to be here:
```
[turtlebot_driver-3] Impossible to communicate with Webots: aborting
[turtlebot_driver-3] Cannot connect to Webots: no pipe defined
[ERROR] [turtlebot_driver-3]: process has died [pid 8889, exit code 1, cmd '/home/matteodr/ros2_webots_ws/install/webots_ros2_turtlebot/lib/webots_ros2_turtlebot/turtlebot_driver --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file /tmp/launch_params_eb0xq060'].
```
Can you see the Webots running?

matteoderose, 2021-05-03:
Exactly. Yes, I can see Webots running. I think the problem is that turtlebot_driver tries to communicate with Webots for 10 seconds, but this time is not sufficient for my Webots to load everything end be up. Is there a point in the source files where I can increase this waiting time?

The same problem occurs with the _universal_robot_ launch file, because the world to load is heavy, while It does not happen with the epuck _robot_launch.py_, since this is a simple world and Webots is able to load everything within 10 secs.

lukicdarkoo, 2021-05-03:
> I  think the problem is that turtlebot_driver tries to communicate with Webots for 10 seconds, but this time is not sufficient for my Webots to load everything and be up.

It is definitely the issue. 

The retry message seems to be a bit confusing, it is actually 55s (the first retry takes 1s, but the second takes 10s). It means that your virtual machine is really slow.

> Is there a point in the source files where I can increase this waiting time?

Unfortunately, there is no way to make it wait for more. I started a PR a while ago to make the controller wait indefinitely, but I didn't manage to make it work properly on Windows:
https://github.com/cyberbotics/webots/pull/2442
More precisely, making the controller wait more is easy, but we had an issue of propagating a kill signal from Python to C on Windows.

Ideas are welcome!

> * Operating System: Linux Ubuntu 20.04 on Virtual Machine through VirtualBox
> * Graphics Card: Intel(R) HD Graphics 4000

You have a very low configuration + you are running everything in a virtual machine without hardware acceleration. Any kind of simulation or visualization will be super slow. Is it possible for you to avoid the virtual machine?

matteoderose, 2021-05-03:
OK thanks, I'll try to do something with my hardware setup and see if there will be any improvements. I'll keep you updated.

lukicdarkoo, 2021-06-01:
@matteoderose Since [this](https://github.com/cyberbotics/webots/pull/2442) is merged I can close the issue. You shouldn't have these problems with R2021b:
https://github.com/cyberbotics/webots/releases

closed 2021-06-01 by lukicdarkoo

matteoderose, 2021-06-01:
Thank you @lukicdarkoo. I finally got an improvement of my setup (using an Intel NUC 8 i7) and all went smoothly.