cyberbotics/webots_ros2 #234: GPS device noise to use in covariance field of NavSatFix.msg

anastasiapan, 2021-06-14:
I have been using Webots for some time now at my work. We have an outdoors platform and we are using gps a lot so I want to simulate that with webots and ros2. I have been able to deploy a gps device successfully (WGS84 in gpsCoordinateSystem) and use it to calculate odometry. What I want to do is add some noise since I am using it for sensor fusion and the covariance is needed.

I am constructing the `NavSatFix` message with the `getValues()` function from `robot.getDevice('gps_device')`. I can see that in the official GPS device documentation (https://cyberbotics.com/doc/reference/gps?tab-language=python) I can change the accuracy and as I understand the number I set to the accuracy is the standard deviation from the gps position in meters. My real life gps device returns an estimation of the horizontal and vertical accuracy. So I calculate the covariance for each measurement. My message looks like that:
```python
class MyNode(WebotsNode):
  self.gps = self.robot.getDevice('gps_device')
  self.gps.enable(32)
  gnss_msg = NavSatFix()
  lla_position = self.gps.getValues()
  gnss_msg.latitude = lla_position[0]
  gnss_msg.longitude = lla_position[1]
  gnss_msg.altitude = lla_position[2]
  # In my real life device I use the squared error. This is what I would like to do with webots too:
  gnss_msg.position_covariance[0] = horizontal_acc**2
  gnss_msg.position_covariance[4] = horizontal_acc**2
  gnss_msg.position_covariance[8] = vertical_acc**2
```

Could I get something like that from webots?

lukicdarkoo, 2021-06-22:
You want to specify horizontal and vertical accuracy in the Webots GPS model? That is not possible in Webots. If necessary we can add, but I think it is fine to add a custom noise (and estimated accuracy) after you get a value from the sensor.

anastasiapan, 2021-06-24:
Thank you for the answer. Horizontal and vertical accuracy is how the GPS devices normally work this is why I mentioned that. I will just add a custom noise then!

lukicdarkoo, 2021-06-24:
Sounds reasonable to have horizontal and vertical accuracy. Please make a feature request to the Webots core:
https://github.com/cyberbotics/webots/issues/new?assignees=&labels=&template=feature_request.md

closed 2021-06-24 by lukicdarkoo

anastasiapan, 2021-06-24:
@lukicdarkoo  I have created the issue https://github.com/cyberbotics/webots/issues/3225 please comment if I have missed something or it is not clear enough