cyberbotics/webots_ros2 #237: Cannot connect to Webots: no pipe defined

Leovilhena, 2021-06-21:
Hi, I'm writing a ROS2 controller almost from scratch and I'm running into issues while launching Webots.

If I launch Webots and then execute ros2 launch... it works but if the launch file starts the Webots simulation I get the following error:

[INFO] [launch]: All log files can be found below /home/leosvilhena/.ros/log/2021-06-21-12-36-06-577080-copacabana-363293
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /dev/null
[INFO] [pioneer_apartment.wbt" --batch --mode=pause-1]: process started with pid [363359]
[INFO] [pioneer_controller-2]: process started with pid [363362]
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 1 second)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 2 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 3 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 4 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 5 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 6 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 7 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 8 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 9 seconds)
[pioneer_controller-2] Webots doesn't seems to be ready yet: (retrying in 10 seconds)
[pioneer_controller-2] Impossible to communicate with Webots: aborting
[pioneer_controller-2] Cannot connect to Webots: no pipe defined
[ERROR] [pioneer_controller-2]: process has died [pid 363362, exit code 1, cmd '/home/leosvilhena/PycharmProjects/pjatk_ros2_ws/install/pioneer/lib/pioneer/pioneer_controller --webots-robot-name Pioneer 3-DX --webots-node-name pioneer __log_level:=debug --clock --ros-args -r __node:=pioneer_3dx --params-file /tmp/launch_params_w2pd32i6'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [pioneer_apartment.wbt" --batch --mode=pause-1]: process has died [pid 363359, exit code -2, cmd '"/home/leosvilhena/.ros/webotsR2021a/webots/webots" "/home/leosvilhena/PycharmProjects/pjatk_ros2_ws/install/pioneer/share/pioneer/worlds/pioneer_apartment.wbt" --batch --mode=pause']

The problem happens exactly in my controller script when I instantiate a Robot object (from controllers import Robot), then it gets stuck and nothing is ran after this statement.
I've tried a lot of combination of parameters but it only works when I starts webots manually first.

Any ideas?

Here is my launch file:

import os
import launch
from launch.substitutions import LaunchConfiguration
from launch.actions import DeclareLaunchArgument, RegisterEventHandler, EmitEvent
from launch import LaunchDescription
from ament_index_python.packages import get_package_share_directory
from webots_ros2_core.utils import ControllerLauncher
from webots_ros2_core.webots_launcher import WebotsLauncher

package_dir = get_package_share_directory('pioneer')

ARGUMENTS = [
    DeclareLaunchArgument(
        'synchronization',
        default_value='True',
        description='If `False` robot.step() will be automatically called.'
    ),
    DeclareLaunchArgument(
        'package',
        default_value='pioneer',
        description='The Package in which the node executable can be found.'
    ),
    DeclareLaunchArgument(
        'executable',
        default_value='pioneer_controller',
        description='The name of the executable to find if a package is provided or otherwise a path to the executable to run.'
    ),
    DeclareLaunchArgument(
        'world',
        default_value=os.path.join(package_dir, 'worlds', 'pioneer_apartment.wbt'),
        description='Path to Webots\' world file. Make sure `controller` field is set to `<extern>`.'
    ),
    DeclareLaunchArgument(
        'gui',
        default_value='True',
        description='Whether to start GUI or not.'
    ),
    DeclareLaunchArgument(
        'mode',
        default_value='pause',
        description='Choose the startup mode (it must be either `pause`, `realtime` or `fast`).'
    ),
    DeclareLaunchArgument(
        'node_parameters',
        description='Path to ROS parameters file that will be passed to the robot node',
        default_value=os.devnull
    ),
    DeclareLaunchArgument(
        'robot_name',
        description='The name of the robot (has to be the same as in Webots)',
        default_value='Pioneer 3-DX'
    ),
    DeclareLaunchArgument(
        'node_name',
        description='The name of the ROS node that interacts with Webots',
        default_value='pioneer'
    ),
    DeclareLaunchArgument(
        'use_sim_time',
        description='Whether to use the simulation (Webots) time',
        default_value='True'
    )
]

def generate_launch_description():
    synchronization = LaunchConfiguration('synchronization')
    package = LaunchConfiguration('package')
    executable = LaunchConfiguration('executable')
    world = LaunchConfiguration('world')
    gui = LaunchConfiguration('gui')
    mode = LaunchConfiguration('mode')
    node_parameters = LaunchConfiguration('node_parameters')
    robot_name = LaunchConfiguration('robot_name')
    node_name = LaunchConfiguration('node_name')
    use_sim_time = LaunchConfiguration('use_sim_time')

    # Webots
    webots = WebotsLauncher(
        world=world,
        mode=mode,
        gui=gui,
    )

    # Driver node
    controller = ControllerLauncher(
        package=package,
        executable=executable,
        name='PIONEER_3DX'.lower(),
        parameters=[
            node_parameters,
            {
                'synchronization': synchronization,
                'user_sim_time': use_sim_time,
            }],
        output='screen',
        arguments=[
            '--webots-robot-name', robot_name,
            '--webots-node-name', node_name,
            '__log_level:=debug',
            '--clock',
        ],
    )

    return LaunchDescription(ARGUMENTS + [
        webots,
        controller,

        # Shutdown launch when Webots exits.
        RegisterEventHandler(
            event_handler=launch.event_handlers.OnProcessExit(
                target_action=webots,
                on_exit=[EmitEvent(event=launch.events.Shutdown())],
            )
        ),
    ])

lukicdarkoo, 2021-06-21:
```
        '--webots-robot-name', robot_name,
```
Does the robot name match the Webots robot name? If you have only one robot in the simulation you can delete that line.

Leovilhena, 2021-06-21:
@lukicdarkoo yes it matches and like I wrote, it works IF I launch Webots manually first.
I tried to remove this line and even though it does not work. Thanks anyway for the reply

Leovilhena, 2021-06-21:
I found the issue. It happens while running the ros2 from Pycharm Terminal. If I do from the normal Ubuntu Terminal it works.

I still don't know why this happens.

lukicdarkoo, 2021-06-21:
Wow, strange, can you run this in the PyCharm terminal:
```
/home/leosvilhena/.ros/webotsR2021a/webots/webots /home/leosvilhena/PycharmProjects/pjatk_ros2_ws/install/pioneer/share/pioneer/worlds/pioneer_apartment.wbt --batch --mode=pause
```

If you figure out why the PyCharm terminal makes the difference then please share it with us

Leovilhena, 2021-06-21:
I'll investigate that, but I'm assuming it might be some permission issue or the fact that Webots is running as a child process of Pycharm.

lukicdarkoo, 2021-06-22:
I will close the issue but feel free to post the findings

closed 2021-06-22 by lukicdarkoo