cyberbotics/webots_ros2 #240: Navigate TurtleBot3

Hanan-abd, 2021-06-30:
Hi I am getting an error [robot_state_publishet-1] : process has died [pid 1378, exit code -6, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args --parms-file tmp/launch_params_wy7t4767']

lukicdarkoo, 2021-06-30:
Can you give me more details, did you compiled it from the source, which Webots version do you use?

There are some weird misspellings, like `robot_state_publishet` and `parms-file`

Hanan-abd, 2021-06-30:
Yes, I compiled it from the source. I completed everything according to these steps.
https://github.com/cyberbotics/webots_ros2/wiki/Navigate-TurtleBot3

Hanan-abd, 2021-06-30:
The system I'm working on is Windows and I used Ubuntu to install ros2-foxy-webots

lukicdarkoo, 2021-06-30:
> I compiled it from the source. 

> I used Ubuntu to install ros2-foxy-webots

You compiled it from the source and installed it through apt? If that is the case you should remove one of them

Hanan-abd, 2021-06-30:
Thanks for your help. but I'm sorry I have a lot of questions
I used webots version 2021a My question is how do I remove apt or webots and what steps do I follow with ubuntu 20 to install ros with webots

closed 2021-07-30 by Hanan-abd