cyberbotics/webots_ros2 #241: urdf-only turtlebot3 simulation

Timple, 2021-07-01:
It would be nice to have a working turtlebot3 simulation example only using urdf.

There has been done a lot of great work on converting urdf (or xacro) to proto files. All seperate elements seem to be present but an integration example seems to be missing.

**Is your feature request related to a problem? Please describe.**
We're trying to setup a webots simulation using urdf including sensors and actuators. But it's quite the flow. And we have to reverse-engineer the urdf2proto conversion to figure out how to get sensors in the proto file.

**Describe the solution you'd like**
A urdf-only implementation for the `webots_ros2_turtlebot`. So basically depend on the `turblebot3_description` package.

**Describe alternatives you've considered**
Currently the only feasible path seems: 
- Add gazebo tags to the urdf so sensors get imported to the proto (seems a bit weird?)
- convert the urdf to a temporary proto file
- Import this proto file to webots (after generation, so launch-order matters)
- Use the ros2_control framework to enable the sensors and actuators (awesome!)

lukicdarkoo, 2021-07-01:
We plan to extend the urdf2webots converter to support `<webots>` tags and to integrate the converter as a `webots_spawner` node.

> convert the urdf to a temporary proto file

We will probably avoid this step and inject nodes directly to the Webots simulation.

> Use the ros2_control framework to enable the sensors and actuators (awesome!)

What do you mean by this? All sensors and actuators should be enabled by default, why would we need the ros2_control framework for that?

Timple, 2021-07-01:
> We plan to extend the urdf2webots converter

Cool, we're starting off on two new robots at the moment. Is there already a branch with some implementation we can test? Or should we start writing proto files?

> What do you mean by this?

My bad. They are enabled. We need ros2_control to get the `/cmd_vel` for our robot.

lukicdarkoo, 2021-07-02:
> Cool, we're starting off on two new robots at the moment. Is there already a branch with some implementation we can test? Or should we start writing proto files?

We haven't started the implementation yet and I am not sure about the schedule, but I will let you know as soon as we start the implementation.

> My bad. They are enabled. We need ros2_control to get the `/cmd_vel` for our robot.

That should work:
https://github.com/cyberbotics/webots_ros2/blob/feedcb5293fc717b41db177eadc56a8b82b50d3c/webots_ros2_turtlebot/launch/robot_launch.py#L57-L68

https://github.com/cyberbotics/webots_ros2/blob/feedcb5293fc717b41db177eadc56a8b82b50d3c/webots_ros2_turtlebot/launch/robot_launch.py#L41-L47

closed 2023-02-20 by ygoumaz