cyberbotics/webots_ros2 #243: libwebots_ros2_driver_imu.so not in lib directory

Timple, 2021-07-12:
**Describe the Bug**
I get this error:
```
[driver-5] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[driver-5]   what():  Could not find library corresponding to plugin webots_ros2_driver::Ros2IMU. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

It seems to be related to `libwebots_ros2_driver_imu.so` not being in the `lib` directory (where the other `.so` files are located, such as `libCppController.so`).

**Steps to Reproduce**
Run the turtlebot demo.

**Workaround**
This seems to work:
```
ln -s "${WS_DIR}/install/webots_ros2_driver/libwebots_ros2_driver_imu.so" "${WS_DIR}/install/webots_ros2_driver/lib/libwebots_ros2_driver_imu.so"
```

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Screenshots**

**System**
 - Webots version: R2021a (snap install)
 - ROS Version: Foxy
 - Operating System: Ubuntu 20.04 derivative

lukicdarkoo, 2021-07-13:
It should be in `install/webots_ros2_driver/lib/`:
```
$ find -name libwebots_ros2_driver_imu.so
./install/webots_ros2_driver/lib/libwebots_ros2_driver_imu.so
$ ls ./install/webots_ros2_driver/lib/
libcar.so  libController.so  libCppCar.so  libCppController.so  libCppDriver.so  libdriver.so  libwebots_ros2_driver_imu.so  webots_ros2_driver
```
see:
https://github.com/cyberbotics/webots_ros2/blob/feedcb5293fc717b41db177eadc56a8b82b50d3c/webots_ros2_driver/CMakeLists.txt#L112-L115

Could you confirm you have pulled the latest version of `webots_ros2`?

Timple, 2021-07-15:
Yes, building from the latest master...

We do however symlink every package to the workspace (so we don't have unused packages in our workspace). And then we use `colcon build --symlink-install` because of https://github.com/colcon/colcon-core/pull/412#issuecomment-856679239

Not sure if that really matters.

lukicdarkoo, 2021-07-15:
You don't need Webots installation to compile the package, it seems you forgot the `--recurse-submodules` flag.
```dockerfile
FROM ros:foxy

# Install webots
RUN apt-get update -qq && apt-get install -qqy software-properties-common wget

# Configure workspace
RUN mkdir -p /ws/src
WORKDIR /ws
RUN apt-get update -qq && apt-get install -qqy python3-colcon-common-extensions

# Install and build (comment one of both methods)
RUN git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2 src/webots_ros2
RUN rosdep install -r --ignore-src --from-paths . || true
RUN bash -c 'source /opt/ros/*/setup.bash && colcon build --packages-up-to webots_ros2_driver'
ENTRYPOINT [ "/usr/bin/find", "-name", "libwebots_ros2_driver_imu.so" ]
```

See:
```bash
$ docker build . --tag webots-docker
$ docker run webots-docker
./build/webots_ros2_driver/libwebots_ros2_driver_imu.so
./install/webots_ros2_driver/lib/libwebots_ros2_driver_imu.so
```

Timple, 2021-07-15:
And here's the problematic version:

```dockerfile
FROM ros:foxy

# Install webots
RUN apt-get update -qq && apt-get install -qqy software-properties-common wget

# Configure workspace
RUN mkdir -p /ws/src
WORKDIR /ws
RUN apt-get update -qq && apt-get install -qqy python3-colcon-common-extensions

# Install and build symlinked
RUN git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2 /webots_ros2
RUN ln -s /webots_ros2/webots_ros2_driver src/webots_ros2_driver
RUN ln -s /webots_ros2/webots_ros2_msgs src/webots_ros2_msgs
RUN rosdep install -ry --ignore-src --from-paths .
RUN bash -c 'source /opt/ros/*/setup.bash && colcon build --symlink-install'
ENTRYPOINT [ "/usr/bin/find", "-name", "libwebots_ros2_driver_imu.so" ]
```

Output:
```
./install/webots_ros2_driver/libwebots_ros2_driver_imu.so
./build/webots_ros2_driver/libwebots_ros2_driver_imu.so
```

It's only the imu though...

```
$ ls install/webots_ros2_driver/
include
lib
libwebots_ros2_driver_imu.so
share

$ ls install/webots_ros2_driver/lib
libController.so
libCppCar.so
libCppController.so
libCppDriver.so
libcar.so
libdriver.so
webots_ros2_driver
```

lukicdarkoo, 2021-07-15:
Oh, I see, the `--symlink-install` flag is the issue. I have to take a look, but the solution should be in CMakeLists.txt

PR #254 (2021-07-23): Fix library directory
**Description**
Fixes #243 

**Related Issues**
#243

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Tasks**
Add the list of tasks of this PR.
  - [x] Confirm for symlinked build
  - [x] Confirm for non-symlinked build
closed 2021-07-23 by Timple