cyberbotics/webots_ros2 #245: Turtlebot3 example can't find world and parameter file

stef264, 2021-07-14:
**Documentation page(s) concerned**
https://github.com/cyberbotics/webots_ros2/wiki/Example-TurtleBot3-Burger

**What is the problem with the current documentation?**
After running the command i get this error:
~~~
stef:~/webots_ws$ ros2 launch webots_ros2_core robot_launch.py \
>     executable:=webots_differential_drive_node \
>     node_parameters:=$(ros2 pkg prefix webots_ros2_examples --share)/resource/turtlebot3_burger.yaml \
>     world:=$(ros2 pkg prefix webots_ros2_examples --share)/worlds/turtlebot3_burger_example.wbt
[INFO] [launch]: All log files can be found below /home/stef/.ros/log/2021-07-14-15-12-46-566654-n-19139
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/stef/webots_ws/install/webots_ros2_examples/share/webots_ros2_examples/resource/turtlebot3_burger.yaml
[INFO] [robot_state_publisher-1]: process started with pid [19142]
[INFO] [turtlebot3_burger_example.wbt" --batch --mode=realtime-2]: process started with pid [19144]
[INFO] [webots_differential_drive_node-3]: process started with pid [19147]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] [INFO] [1626268366.639534795] [robot_state_publisher]: got segment 
[turtlebot3_burger_example.wbt" --batch --mode=realtime-2] Could not open file: '/home/stef/webots_ws/install/webots_ros2_examples/share/webots_ros2_examples/worlds/turtlebot3_burger_example.wbt'.
[webots_differential_drive_node-3] Cannot open file: /tmp/webots-19146-33RMSF/WEBOTS_SERVER (retrying in 1 second)
[INFO] [turtlebot3_burger_example.wbt" --batch --mode=realtime-2]: process has finished cleanly [pid 19144]
[INFO] [webots_differential_drive_node-3]: sending signal 'SIGINT' to process[webots_differential_drive_node-3]
[INFO] [robot_state_publisher-1]: sending signal 'SIGINT' to process[robot_state_publisher-1]
[webots_differential_drive_node-3] Cannot open file: /tmp/webots-19146-33RMSF/WEBOTS_SERVER (retrying in 2 seconds)
[robot_state_publisher-1] [INFO] [1626268367.298959643] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 19142]
[webots_differential_drive_node-3] Cannot open file: /tmp/webots-19146-33RMSF/WEBOTS_SERVER (retrying in 3 seconds)
[webots_differential_drive_node-3] Cannot open file: /tmp/webots-19146-33RMSF/WEBOTS_SERVER (retrying in 4 seconds)
[ERROR] [webots_differential_drive_node-3]: process[webots_differential_drive_node-3] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [webots_differential_drive_node-3]: sending signal 'SIGTERM' to process[webots_differential_drive_node-3]
[ERROR] [webots_differential_drive_node-3]: process has died [pid 19147, exit code -15, cmd '/home/stef/webots_ws/install/webots_ros2_core/lib/webots_ros2_core/webots_differential_drive_node --webots-robot-name  --webots-node-name webots_driver --ros-args --params-file /tmp/launch_params_hnd8r5tf'].
~~~

I'm using Ubuntu 20.04 + ros2 foxy

i can find a world file at:
~~~
$(ros2 pkg prefix webots_ros2_turtlebot --share)/worlds/turtlebot3_burger_example.wbt
~~~
instead of
~~~
$(ros2 pkg prefix webots_ros2_examples --share)/worlds/turtlebot3_burger_example.wbt
~~~

But the parameter yaml file is missing.

Is this intended? Should i create my own world and parameter file?

lukicdarkoo, 2021-07-14:
I have just updated the documentation page, please try simply running:
```
ros2 launch webots_ros2_turtlebot robot_launch.py
```

stef264, 2021-07-14:
It works. Thank you!

closed 2021-07-14 by stef264