cyberbotics/webots_ros2 #256: Failed   <<< webots_ros2_driver [9.47s, exited with code 1]

zhangrelay, 2021-07-27:
**Describe the Bug**
win10 webots_ros2 1.1.0 foxy with webots 2021b

C:\ros_ws\webots_ros2>colcon build
[1.556s] root DEBUG Using proactor: IocpProactor
Starting >>> vision_msgs
Starting >>> webots_ros2_ur_e_description
Starting >>> webots_ros2_importer
--- stderr: webots_ros2_importer
error: package directory 'webots_ros2_importer\urdf2webots\urdf2webots' does not exist
---
Failed   <<< webots_ros2_importer [3.47s, exited with code 1]
Aborted  <<< webots_ros2_ur_e_description [3.92s]
Aborted  <<< vision_msgs [13.7s]

Summary: 0 packages finished [14.3s]
  1 package failed: webots_ros2_importer
  2 packages aborted: vision_msgs webots_ros2_ur_e_description
  1 package had stderr output: webots_ros2_importer
  15 packages not processed

C:\ros_ws\webots_ros2>colcon build
[1.571s] root DEBUG Using proactor: IocpProactor
Starting >>> vision_msgs
Starting >>> webots_ros2_ur_e_description
Starting >>> webots_ros2_importer
Finished <<< webots_ros2_ur_e_description [3.91s]
Finished <<< webots_ros2_importer [3.83s]
[Processing: vision_msgs]vision_msgs:build - 34.0s]
Finished <<< vision_msgs [48.8s]
Starting >>> webots_ros2_msgs
[Processing: webots_ros2_msgs]
Finished <<< webots_ros2_msgs [34.1s]
Starting >>> webots_ros2_core
Starting >>> webots_ros2_driver
Finished <<< webots_ros2_core [3.62s]
Starting >>> webots_ros2_abb
Starting >>> webots_ros2_universal_robot
Starting >>> webots_ros2_epuck
Starting >>> webots_ros2_examples
Starting >>> webots_ros2_mavic
Starting >>> webots_ros2_tesla
Starting >>> webots_ros2_tiago
Finished <<< webots_ros2_abb [12.7s]
Starting >>> webots_ros2_tutorials
Finished <<< webots_ros2_universal_robot [12.9s]
Starting >>> webots_ros2_demos
Finished <<< webots_ros2_tiago [12.1s]
Finished <<< webots_ros2_mavic [12.7s]
Finished <<< webots_ros2_examples [13.0s]
Finished <<< webots_ros2_tesla [12.6s]
Finished <<< webots_ros2_epuck [13.3s]
Finished <<< webots_ros2_tutorials [5.91s]
Finished <<< webots_ros2_demos [8.31s]
Failed   <<< webots_ros2_driver [43.9s, exited with code 1]

Summary: 14 packages finished [2min 7s]
  1 package failed: webots_ros2_driver
  3 packages not processed

**Affected Packages**
List of affected packages:
  - webots_ros2_driver

**Screenshots**

[image]

`
C:\ros_ws\webots_ros2>colcon build --packages-up-to webots_ros2_turtlebot
[1.491s] root DEBUG Using proactor: IocpProactor
Starting >>> vision_msgs
Finished <<< vision_msgs [11.9s]
Starting >>> webots_ros2_msgs
Finished <<< webots_ros2_msgs [11.6s]
Starting >>> webots_ros2_driver
Starting >>> webots_ros2_core
Finished <<< webots_ros2_core [3.78s]
Failed   <<< webots_ros2_driver [9.47s, exited with code 1]

Summary: 3 packages finished [33.4s]
  1 package failed: webots_ros2_driver
  2 packages not processed
`

**System**
 - Webots Version: [2021b]
 - ROS Version: [foxy] 
 - Operating System: [Windows 10]
 - Graphics Card: [Intel]

**Additional context**
Add any other context about the problem here.

lukicdarkoo, 2021-07-27:
Thank you for reporting the problem, we working on fixing it. Meanwhile, if you are in hurry then you checkout at ad78a04660eccead35f9dd0

zhangrelay, 2021-07-28:
> Thank you for reporting the problem, we working on fixing it. Meanwhile, if you are in hurry then you checkout at [ad78a04](https://github.com/cyberbotics/webots_ros2/commit/ad78a04660eccead35f9dd07d1c31a0d5cd4c2ad)

Thank you very much, let me give it a try.

zhangrelay, 2021-07-28:
> Thank you for reporting the problem, we working on fixing it. Meanwhile, if you are in hurry then you checkout at [ad78a04](https://github.com/cyberbotics/webots_ros2/commit/ad78a04660eccead35f9dd07d1c31a0d5cd4c2ad)

Finished <<< webots_ros2_driver [20.4s]  This is ok.
But there is one last little problem.

```
Failed   <<< webots_ros2_control [11.6s, exited with code 1]

Summary: 16 packages finished [1min 4s]
  1 package failed: webots_ros2_control
  2 packages not processed
```

Looking forward to the 1.1.1 release.

Thanks a lot for reply.

lukicdarkoo, 2021-07-28:
A few questions:
- Did you try from https://github.com/cyberbotics/webots_ros2/pull/258?
- Can you use the `--event-handlers console_cohesion+` flag so we get more details (e.g. `colcon build --event-handlers console_cohesion+`)?
- Did you sync the submodules, `git pull --recurse-submodules`?

zhangrelay, 2021-07-29:
> A few questions:
> 
> * Did you try from [Fix Windows build #258](https://github.com/cyberbotics/webots_ros2/pull/258)?
> * Can you use the `--event-handlers console_cohesion+` flag so we get more details (e.g. `colcon build --event-handlers console_cohesion+`)?
> * Did you sync the submodules, `git pull --recurse-submodules`?
> * Do you have necessary dependencies:
>   ```
>   git clone --branch=ros2 https://github.com/ros-perception/vision_msgs.git src/vision_msgs
>   git clone --branch=ros2 https://github.com/ros-drivers/ackermann_msgs.git src/ackermann_msgs
>   git clone https://github.com/ros-controls/ros2_control.git src/ros2_control
>   git clone https://github.com/ros-controls/ros2_controllers.git src/ros2_controllers
>   git clone --branch=foxy-devel https://github.com/ros-controls/control_msgs.git src/control_msgs
>   git clone --branch=ros2 https://github.com/ros/angles.git src/angles
>   git clone --branch=foxy-devel https://github.com/ros-controls/realtime_tools.git src/realtime_tools
>   ```

I followed the steps to compile and it was completely fine.

1. git clone --branch=fix-windows-build --recurse-submodules  https://github.com/cyberbotics/webots_ros2.git
2. git all necessary dependencies
3. colcon build --event-handlers console_cohesion+

Finished <<< webots_ros2 [7.98s]

Summary: 42 packages finished [2min 57s]

colcon build is OK.

zhangrelay, 2021-07-29:
Now all webots_ros2 demos can be run and used normally in win10 + ROS2.

closed 2021-07-29 by zhangrelay