cyberbotics/webots_ros2 #260: Why are we using rclcpp clock instead of the ros2 node clock?

stef264, 2021-08-09:
By running the basic turtlebot 3 demo, I noticed the LiDAR scans getting published with my system real-time clock instead of sim time. Shouldn't it respect `use_sim_time = true` and publish in sim time?

I changed this line ...
https://github.com/cyberbotics/webots_ros2/blob/9bd50fb01c3de3704664fc3b41d00e3600a00fb6/webots_ros2_driver/src/plugins/static/Ros2Lidar.cpp#L132

.. to
```c++
auto clock = node->get_clock();
mLaserMessage.header.stamp = clock->now();
```

and copying a reference `node` from webots_ros2_driver::WebotsNode *node in the init function.

Is this intended, or am I misunderstanding the ros2 time concept?

lukicdarkoo, 2021-08-09:
I believe we should use this way instead: https://github.com/cyberbotics/webots_ros2/blob/9bd50fb01c3de3704664fc3b41d00e3600a00fb6/webots_ros2_core/webots_ros2_core/devices/sensor_device.py#L66

since `use_sim_time = false` wouldn't make much sense with Webots. Or I am assuming something wrong?

stef264, 2021-08-09:
I don't really understand how the linked line let the lidar use the node clock.

output of: `ros2 topic echo /scan` (on master)
```
---
header:
  stamp:
    sec: 1628546446
    nanosec: 876792030
  frame_id: LDS_custom_rotated
angle_min: -3.141590118408203
angle_max: 3.141590118408203
angle_increment: 0.017453277483582497
time_increment: 0.0
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
ranges:
... SCAN DATA
intensities: []
---
```

Shouldn't the expected output be this? (using my code changes)
```
---
header:
  stamp:
    sec: 6
    nanosec: 990000000
  frame_id: LDS_custom_rotated
angle_min: -3.141590118408203
angle_max: 3.141590118408203
angle_increment: 0.017453277483582497
time_increment: 0.0
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
ranges:
... SCAN DATA
intensities: []
---
```

The Lidar device isn't the only one using the rclcpp clock for timestamping.
I'm pretty new to ROS and Webots, so i don't know if this is a bug.

Timple, 2021-08-10:
> since use_sim_time = false wouldn't make much sense with Webots.

Well, theoretically: If the PC is fast enough to simulate realtime (1x), you could use hardware-in-the-loop for example. And thus not use sim_time.

lukicdarkoo, 2021-08-10:
> I don't really understand how the linked line let the lidar use the node clock.

It doesn't, it uses the simulated time from Webots. The effect should be the same. Since we are in the simulation I thought the real-time clock wouldn't be needed.

However, since @Timple has suggested a theoretical use-case we can go with your option (node clock). Can you make a PR?

closed 2021-08-11 by stef264