cyberbotics/webots_ros2 #261: Navigate TurtleBot3 Issue

MahouBootCamp, 2021-08-10:
I'm trying the navigation demo and cannot get the navigation system work properly. It's my first time using Webots so I just follow the instruction and install everything by package manager. Webots is R2021b and is downloaded automatically.

The webots and rviz gui show up, but there's no map topic and no transform from /odom to /map/

From log what I can see it fails to load plugin `webots_ros2_driver::Ros2IMU`(whose .so file actually exists) and then `/controller_manager` fails to run causing `spawner_diffdrive_controller` & `spawner_joint_state_broadcaster`'s failure.

Detailed log is listed below.

Terminal output of webots:
```
[INFO] [launch]: All log files can be found below /home/mhbc/.ros/log/2021-08-10-11-51-01-246711-mhbc-ROS-25904
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [spawner.py-1]: process started with pid [25908]
[INFO] [spawner.py-2]: process started with pid [25910]
[INFO] [turtlebot3_burger_example.wbt" --batch --mode=realtime-3]: process started with pid [25913]
[INFO] [robot_state_publisher-4]: process started with pid [25915]
[INFO] [driver-5]: process started with pid [25919]
[INFO] [static_transform_publisher-6]: process started with pid [25921]
[driver-5] Cannot open file: /tmp/webots-25918-2iixXa/WEBOTS_SERVER (retry count 1)
[static_transform_publisher-6] [INFO] [1628567461.333937935] [static_transform_publisher_2vigRTd6kwzljIGM]: Spinning until killed publishing transform from 'base_link' to 'base_footprint'
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] [INFO] [1628567461.335604475] [robot_state_publisher]: got segment 
[driver-5] Cannot open file: /tmp/webots-25918-2iixXa/WEBOTS_SERVER (retry count 2)
[driver-5] Cannot open file: /tmp/webots-25918-2iixXa/WEBOTS_SERVER (retry count 3)
[driver-5] Cannot open file: /tmp/webots-25918-2iixXa/WEBOTS_SERVER (retry count 4)
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link LDS-01 had 6 children
[robot_state_publisher-4] Link solid(1) had 0 children
[robot_state_publisher-4] Link screw0 had 0 children
[robot_state_publisher-4] Link screw1 had 0 children
[robot_state_publisher-4] Link screw2 had 0 children
[robot_state_publisher-4] Link screw3 had 0 children
[robot_state_publisher-4] Link solid_0 had 0 children
[robot_state_publisher-4] Link accelerometer had 0 children
[robot_state_publisher-4] Link compass had 0 children
[robot_state_publisher-4] Link gyro had 0 children
[robot_state_publisher-4] Link imu_link had 1 children
[robot_state_publisher-4] Link solid had 0 children
[robot_state_publisher-4] Link inertial_unit had 0 children
[robot_state_publisher-4] Link left wheel had 0 children
[robot_state_publisher-4] Link right wheel had 0 children
[robot_state_publisher-4] [INFO] [1628567465.681053929] [robot_state_publisher]: got segment LDS-01
[robot_state_publisher-4] [INFO] [1628567465.681093664] [robot_state_publisher]: got segment accelerometer
[robot_state_publisher-4] [INFO] [1628567465.681101769] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1628567465.681108703] [robot_state_publisher]: got segment compass
[robot_state_publisher-4] [INFO] [1628567465.681115355] [robot_state_publisher]: got segment gyro
[robot_state_publisher-4] [INFO] [1628567465.681122278] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1628567465.681128921] [robot_state_publisher]: got segment inertial_unit
[robot_state_publisher-4] [INFO] [1628567465.681135443] [robot_state_publisher]: got segment left wheel
[robot_state_publisher-4] [INFO] [1628567465.681142105] [robot_state_publisher]: got segment right wheel
[robot_state_publisher-4] [INFO] [1628567465.681148638] [robot_state_publisher]: got segment screw0
[robot_state_publisher-4] [INFO] [1628567465.681155120] [robot_state_publisher]: got segment screw1
[robot_state_publisher-4] [INFO] [1628567465.681161562] [robot_state_publisher]: got segment screw2
[robot_state_publisher-4] [INFO] [1628567465.681168024] [robot_state_publisher]: got segment screw3
[robot_state_publisher-4] [INFO] [1628567465.681174637] [robot_state_publisher]: got segment solid
[robot_state_publisher-4] [INFO] [1628567465.681181089] [robot_state_publisher]: got segment solid(1)
[robot_state_publisher-4] [INFO] [1628567465.681189164] [robot_state_publisher]: got segment solid_0
[driver-5] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[driver-5]   what():  Could not find library corresponding to plugin webots_ros2_driver::Ros2IMU. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [driver-5]: process has died [pid 25919, exit code -6, cmd '/opt/ros/foxy/lib/webots_ros2_driver/driver --ros-args --params-file /tmp/launch_params_si1wvkcb --params-file /opt/ros/foxy/share/webots_ros2_turtlebot/resource/ros2control.yml -r /diffdrive_controller/cmd_vel_unstamped:=/cmd_vel'].
[spawner.py-2] [INFO] [1628567471.443953622] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1628567471.450646803] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-2] [INFO] [1628567473.451271959] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1628567473.458582122] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-2] [INFO] [1628567475.458691158] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1628567475.465675046] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-2] [INFO] [1628567477.466440402] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1628567477.472865536] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-2] [INFO] [1628567479.473855647] [spawner_diffdrive_controller]: Waiting for /controller_manager services
[spawner.py-1] [INFO] [1628567479.480600205] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[... 316 lines cut ...]
[amcl-2] [INFO] [1628567506.467169009] [rclcpp]: signal_handler(signal_value=2)
[amcl-2] [INFO] [1628567506.471245308] [amcl]: Destroying
[recoveries_server-6] [INFO] [1628567506.467167235] [rclcpp]: signal_handler(signal_value=2)
[planner_server-5] [INFO] [1628567506.467168658] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-3] [INFO] [1628567506.467169309] [rclcpp]: signal_handler(signal_value=2)
[controller_server-4] [INFO] [1628567506.467172305] [rclcpp]: signal_handler(signal_value=2)
[controller_server-4] [INFO] [1628567506.467384925] [local_costmap.local_costmap]: start
[controller_server-4] [ERROR] [1628567506.467584231] []: Caught exception in callback for transition 13
[controller_server-4] [ERROR] [1628567506.467598167] []: Original error: could not create subscription: rcl node's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/node.c:441
[controller_server-4] [ERROR] [1628567506.467621311] []: Failed to finish transition 3. Current state is now: errorprocessing (Could not publish transition: publisher's context is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/publisher.c:423, at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.11/src/rcl_lifecycle.c:367)
[map_server-1] [INFO] [1628567506.467177835] [rclcpp]: signal_handler(signal_value=2)
[map_server-1] [INFO] [1628567506.468282136] [map_server]: Destroying
[bt_navigator-7] [PublisherZMQ] Server quitting.
[bt_navigator-7] [PublisherZMQ] just died. Exeption Context was terminated
[bt_navigator-7] [PublisherZMQ] Publisher quitting.
[bt_navigator-7] [PublisherZMQ] just died. Exeption Context was terminated
[lifecycle_manager-3] [INFO] [1628567506.476703310] [lifecycle_manager_localization]: Destroying
[INFO] [map_server-1]: process has finished cleanly [pid 26133]
[INFO] [waypoint_follower-8]: process has finished cleanly [pid 26147]
[INFO] [lifecycle_manager-3]: process has finished cleanly [pid 26137]
[INFO] [amcl-2]: process has finished cleanly [pid 26135]
[INFO] [bt_navigator-7]: process has finished cleanly [pid 26145]
[ERROR] [rviz2-10]: process has died [pid 26151, exit code -6, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /opt/ros/foxy/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2 -r __node:=rviz2 --params-file /tmp/launch_params_symx73ri'].
[ERROR] [lifecycle_manager-9]: process has died [pid 26149, exit code -6, cmd '/opt/ros/foxy/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args -r __node:=lifecycle_manager_navigation --params-file /tmp/launch_params_1z45enyy --params-file /tmp/launch_params_2wu68eo5 --params-file /tmp/launch_params_cdjg_v_b'].
```

lukicdarkoo, 2021-08-10:
It seems to be a bug when packaging. Thank you for reporting the issue.

Until we fix it could you please compile it from the source:
https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install

lukicdarkoo, 2021-08-10:
The `libwebots_ros2_driver_imu.so` library ended up in the `/opt/ros/foxy/lib/x86_64-linux-gnu` directory instead of `/opt/ros/foxy/lib`. That's way it cannot be found by `pluginlib`. Other dynamic libraries (e.g. webots_ros2_control) are properly placed in the `/opt/ros/foxy/lib` directory.

closed 2021-08-11 by lukicdarkoo

MahouBootCamp, 2021-08-12:
> It seems to be a bug when packaging. Thank you for reporting the issue.
> 
> Until we fix it could you please compile it from the source:
> https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install

Thanks! That works for me!

lukicdarkoo, 2021-09-04:
@MahouBootCamp The update is synced, you can install it through apt:
https://discourse.ros.org/t/new-packages-and-patch-release-for-foxy-fitzroy-2021-09-02/22139

PR #262 (2021-08-11): Fix the IMU library exportation
This PR should resolve #261