cyberbotics/webots_ros2 #267: Cannot find tinyxml2&vision_msgs&control_msgs&ros2_control

dzywater, 2021-08-15:
1.Environment
system: Windows 10
foxy: release4 (ros2-foxy-20201211-windows-release.amd64.zip)
webots_ros2: b2e7635

2.When compiled, cannot find: tinyxml2、vision_msgs、control_msgs、ros2_control

3.I solved it: 
(1)Download&compile&install: https://github.com/leethomason/tinyxml2.git   (9.0.0)
(2)Download: https://github.com/ros-perception/vision_msgs.git  (3.0.1) and compile with webots_ros2
(3)Download: https://github.com/ros-controls/control_msgs (3.0.0)
(4)Download: https://github.com/ros-controls/ros2_control  (108671d) and compile with webots_ros2
note1: latest version 0.7.1 is not workable
noet2: add M_PI defination in ros2_control\hardware_interface\test\test_component_parser.cpp

So, .gitmodules or something needs to be udpated and

lukicdarkoo, 2021-08-15:
We cannot put these packages as submodules. Those packages are available in ROS 2 distributions and we should use standard ROS installation methods.

Ideally, those packages should be installed with `rosinstall_generator`, but I couldn't make it work properly. It seems there are some bugs and it doesn't work as nicely as on Ubuntu. I may give it a second try.

The other way is to document a list of dependencies (like in [this comment](https://github.com/cyberbotics/webots_ros2/issues/256#issuecomment-888052221)), but it is cumbersome to maintain the list for multiple distributions.

dzywater, 2021-08-15:
My ideas:
1.For Ubuntu, detect whether all dependencies exist or not. If not, apt-get install *** 
2.For Windows, detect whether all dependencies exist or not. If not, git clone ** and compile with webots_ros2 together.

lukicdarkoo, 2021-08-15:
> 1.For Ubuntu, detect whether all dependencies exist or not. If not, apt-get install ***

For Ubuntu is easy.  In the documentation, we suggest using the `rosdep` tool to install dependencies, see [here](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#linux)

> 2.For Windows, detect whether all dependencies exist or not. If not, git clone ** and compile with webots_ros2 together.

Windows is tricky, `rosdep` doesn't work on Windows. Therefore, the sources have to be manually managed. The `rosinstall_generator` tool should do the job, but in my tests, it was either downloading unnecessary packages or none. I will try to see whats the exact problem is and post a solution.

lukicdarkoo, 2021-08-25:
I have just added a Dependencies section:
https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#dependencies

dzywater, 2021-08-29:
> I have just added a Dependencies section:
> https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#dependencies

It's great!

closed 2021-08-29 by dzywater

dzywater, 2021-08-29:
> I have just added a Dependencies section:
> https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install#dependencies

Until now, these packages "ackermann_msgs  ros2_controllers angles realtime_tools" are not  used when compling webots_ros2, but it succeeded. I don't know why.