cyberbotics/webots_ros2 #276: It may be more efficient and consistent o use "device" param in getDeviceRosProperties function ???

dzywater, 2021-08-25:
To find a plugin, use:
   mWebotsXMLElement->FirstChildElement("plugin");
   pluginElement = pluginElement->NextSiblingElement("plugin");

So, to find a webots device, use:
   mWebotsXMLElement->FirstChildElement("device") instead of mWebotsXMLElement->FirstChildElement() in line89
   deviceChild = deviceChild->NextSiblingElement("device") instead of mWebotsXMLElement->FirstChildElement() in line94

lukicdarkoo, 2021-08-25:
The efficiency is not a concern in that part of the code, the configuration files are tiny, so the startup time will be almost the same. Error handling is important though.

lukicdarkoo, 2021-12-21:
I believe we can close this issue

closed 2021-12-21 by lukicdarkoo