cyberbotics/webots_ros2 #288: New flake8 and pep257 errors

BenjaminHug8, 2021-08-31:
**Description of the Bug**
By running "colon test" after a recent installation of ROS2, several tests will fail due to flake8 and pep257 conventions.

**Steps to Reproduce**
1. Install ROS2 Foxy
2. Install webots_ros2 like [here](https://github.com/cyberbotics/webots_ros2/wiki/Build-and-Install)
3. Checkout on feature-deprecate-core
4. Run "colon test"
5. Run "colon test-result" for more explanation

**Expected behavior**
By running "colon test" after a recent installation of ROS2, 14 failures contained in 13 packages over the 351 tests will rise.

**Affected Packages**
List of affected packages with the tests run on them:

- build/webots_ros2_abb/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_core/pytest.xml: 9 tests, 0 errors, 2 failures, 0 skipped
- build/webots_ros2_demos/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_epuck/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_examples/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_importer/pytest.xml: 4 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_mavic/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_tesla/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_tests/pytest.xml: 6 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_tiago/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_turtlebot/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_tutorials/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- build/webots_ros2_universal_robot/pytest.xml: 3 tests, 0 errors, 1 failure, 0 skipped

lukicdarkoo, 2021-08-31:
I get only one:
```
$ colcon test
...
Summary: 19 packages finished [1min 9s]
  1 package had test failures: webots_ros2_core
```

More details about the failed test:
```xml
<testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="52" name="test_lookup_asc_edge" time="0.001"></testcase><testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="56" name="test_lookup_asc_extrapolation" time="0.000"></testcase><testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="48" name="test_lookup_asc_interpolation" time="0.000"></testcase><testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="40" name="test_lookup_desc_edge" time="0.000"></testcase><testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="44" name="test_lookup_desc_extrapolation" time="0.000"></testcase><testcase classname="webots_ros2_core.test.test_math_utils.TestMathUtils" file="test/test_math_utils.py" line="36" name="test_lookup_desc_interpolation" time="0.001"></testcase><testcase classname="webots_ros2_core.test.test_pep257" file="test/test_pep257.py" line="20" name="test_pep257" time="0.192"><failure message="AssertionError: Found code style errors / warnings
assert 1 == 0">test/test_pep257.py:25: in test_pep257
    assert rc == 0, &apos;Found code style errors / warnings&apos;
E   AssertionError: Found code style errors / warnings
E   assert 1 == 0</failure><system-out>checking ./launch/robot_launch.py
checking ./setup.py
checking ./test/test_copyright.py
checking ./test/test_flake8.py
checking ./test/test_math_utils.py
checking ./test/test_pep257.py
checking ./webots_ros2_core/__init__.py
checking ./webots_ros2_core/devices/__init__.py
checking ./webots_ros2_core/devices/camera_device.py
./webots_ros2_core/devices/camera_device.py:29 in public class `CameraDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/device.py
checking ./webots_ros2_core/devices/device_manager.py
checking ./webots_ros2_core/devices/distance_sensor_device.py
./webots_ros2_core/devices/distance_sensor_device.py:24 in public class `DistanceSensorDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/gps_device.py
./webots_ros2_core/devices/gps_device.py:26 in public class `GpsDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/imu_device.py
./webots_ros2_core/devices/imu_device.py:24 in public class `ImuDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/led_device.py
./webots_ros2_core/devices/led_device.py:23 in public class `LEDDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/lidar_device.py
./webots_ros2_core/devices/lidar_device.py:25 in public class `LidarDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/light_sensor_device.py
./webots_ros2_core/devices/light_sensor_device.py:28 in public class `LightSensorDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/range_finder_device.py
./webots_ros2_core/devices/range_finder_device.py:24 in public class `RangeFinderDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/robot_device.py
./webots_ros2_core/devices/robot_device.py:28 in public class `RobotDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/devices/sensor_device.py
./webots_ros2_core/devices/sensor_device.py:22 in public class `SensorDevice`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/joint_state_publisher.py
./webots_ros2_core/joint_state_publisher.py:23 in public class `JointStatePublisher`: D413: Missing blank line after last section (&apos;Args&apos;)
./webots_ros2_core/joint_state_publisher.py:23 in public class `JointStatePublisher`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/math/__init__.py
checking ./webots_ros2_core/math/interpolation.py
checking ./webots_ros2_core/trajectory_follower.py
checking ./webots_ros2_core/utils.py
checking ./webots_ros2_core/webots/__init__.py
checking ./webots_ros2_core/webots/controller.py
checking ./webots_ros2_core/webots/vehicle.py
checking ./webots_ros2_core/webots_differential_drive_node.py
./webots_ros2_core/webots_differential_drive_node.py:29 in public class `WebotsDifferentialDriveNode`: D413: Missing blank line after last section (&apos;Args&apos;)
./webots_ros2_core/webots_differential_drive_node.py:29 in public class `WebotsDifferentialDriveNode`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/webots_launcher.py
checking ./webots_ros2_core/webots_node.py
./webots_ros2_core/webots_node.py:38 in public class `WebotsNode`: D413: Missing blank line after last section (&apos;Args&apos;)
./webots_ros2_core/webots_node.py:38 in public class `WebotsNode`: D407: Missing dashed underline after section (&apos;Args&apos;)
checking ./webots_ros2_core/webots_robotic_arm_node.py
./webots_ros2_core/webots_robotic_arm_node.py:23 in public class `WebotsRoboticArmNode`: D413: Missing blank line after last section (&apos;Args&apos;)
./webots_ros2_core/webots_robotic_arm_node.py:23 in public class `WebotsRoboticArmNode`: D407: Missing dashed underline after section (&apos;Args&apos;)
18 errors
</system-out></testcase></testsuite></testsuites>
```

BenjaminHug8, 2021-08-31:
Might be due to different versions of flake8.

What is yours ?

Mine is 3.7.9-2

omichel, 2021-08-31:
@BenjaminHug8: Beware, you are misspelling `colcon`.

lukicdarkoo, 2021-08-31:
> Might be due to different versions of flake8.
> 
> What is yours ?
> 
> Mine is 3.7.9-2

```
$ flake8 --version
3.8.4 (mccabe: 0.6.1, pycodestyle: 2.6.0, pyflakes: 2.2.0) CPython 3.8.10 on
Linux
```

BenjaminHug8, 2021-09-01:
> @BenjaminHug8: Beware, you are misspelling `colcon`.

Thank you Olivier, I fixed it.

BenjaminHug8, 2021-09-01:
As discussed with @lukicdarkoo, we can't reproduce the errors of flake8 convention and will focus only on pep257 error.

Thus I change the description and title of the issue

closed 2021-09-01 by BenjaminHug8