cyberbotics/webots_ros2 #295: Galactic - missing dependencies

anastasiapan, 2021-09-03:
**Describe the Bug**
I want to update webots_ros2 to galactic but there are some dependencies that do not exist yet for galactic so the build fails (controller_manager, joint_state_broadcaster)

**Steps to Reproduce**
1. Checkout galactic branch
2. `rosdep update`
3. `rosdep install --from-paths src --ignore-src --rosdistro galactic`
4. ```Cannot locate rosdep definition for [controller_manager]
         Cannot locate rosdep definition for [joint_state_broadcaster]```

**Expected behavior**
Install the dependencies

**Affected Packages**
List of affected packages:
  - webots_ros2_control
  - webots_ros2_turtlebot

**System**
 - Webots Version: 2021b
 - ROS Version: Galactic 
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 2060

lukicdarkoo, 2021-09-03:
Unfortunately, the mentioned packages are not released yet. It is a known problem and it is blocking us from releasing new versions to Galactic:
https://github.com/cyberbotics/webots_ros2/pull/250

You can switch to Rolling or Foxy. Or, simply ignore `webots_ros2_control` and `webots_ros2_turtlebot`, the other packages are not affected.

anastasiapan, 2021-09-03:
@lukicdarkoo  I think I will just ignore the packages for now I am not using them.

closed 2021-09-03 by anastasiapan