cyberbotics/webots_ros2 #297: Why define gyro's data as acceleration and gps's second part as altitude

dzywater, 2021-09-04:
https://github.com/cyberbotics/webots_ros2/blob/f5f33a9035bf3a1152a9f9df4e1001d5db3375dd/webots_ros2_mavic/webots_ros2_mavic/mavic_driver.py#L82
and 
https://github.com/cyberbotics/webots/blob/5c3c33470190fad062591e6f681c7967982c070f/projects/robots/dji/mavic/controllers/mavic2pro/mavic2pro.c#L116
define gyro's data as acceleration, but actually velocity, as described at https://www.cyberbotics.com/doc/reference/gyro?tab-language=c#wb_gyro_get_values

https://github.com/cyberbotics/webots/blob/5c3c33470190fad062591e6f681c7967982c070f/projects/robots/dji/mavic/controllers/mavic2pro/mavic2pro.c#L115
define gps's second part as altitude, but longitude, as described at https://www.cyberbotics.com/doc/reference/gps?tab-language=c#wb_gps_get_values

lukicdarkoo, 2021-09-14:
> https://github.com/cyberbotics/webots/blob/5c3c33470190fad062591e6f681c7967982c070f/projects/robots/dji/mavic/controllers/mavic2pro/mavic2pro.c #L115
> define gps's second part as altitude, but longitude, as descriped at https://www.cyberbotics.com/doc/reference/gps?tab-language=c#wb_gps_get_values

In ENU, the third value is altitude.

> https://github.com/cyberbotics/webots_ros2/blob/f5f33a9035bf3a1152a9f9df4e1001d5db3375dd/webots_ros2_mavic/webots_ros2_mavic/mavic_driver.py #L82
> and
> https://github.com/cyberbotics/webots/blob/5c3c33470190fad062591e6f681c7967982c070f/projects/robots/dji/mavic/controllers/mavic2pro/mavic2pro.c #L116
> define gyro's data as acceleration, but actually velocity, as descriped at https://www.cyberbotics.com/doc/reference/gyro?tab-language=c#wb_gyro_get_values

This is inherited from https://github.com/cyberbotics/webots/blob/master/projects/robots/dji/mavic/controllers/mavic2pro/mavic2pro.c, we should probably fix it.

dzywater, 2021-09-14:
@lukicdarkoo , I suggested to add webots simulation on https://github.com/PX4/PX4-Autopilot/issues/18221. Do you have any advice ?

lukicdarkoo, 2021-09-14:
> @lukicdarkoo , I suggested to add webots simulation on [PX4/PX4-Autopilot#18221](https://github.com/PX4/PX4-Autopilot/issues/18221). Do you have any advice ?

That would be lovely, but I am not sure whether it's our priority right now. Could you please create an issue in the `webots` repository?

BTW, there is already an ArduPilot integration:
https://ardupilot.org/dev/docs/sitl-with-webots.html

lukicdarkoo, 2021-09-14:
Does PX4 depend on ROS? I see some launch files there, but I thought PX4 is standalone.

dzywater, 2021-09-15:
PX4 is standalone.
It can communicate with  the external by 
	MAVLink protocol, e.g MAVSDK(a high-level API to MAVLink)
	MicroRTPS, a type of DDS
	ROS or ROS2, but this way is still by MicroRTPS
	MAVROS, similar with webots_ros2 principle 
MAVLink  and MAVROS are general and can be used for PX4 and ArduPilot.
See: https://github.com/mavlink/mavros/issues/1369
I knew ArduPilot webots simulation.
I aslo want to have PX4 webots simulatio.

PR #636 (2023-02-21): Rename gyro variables to velocity
Fixes #297.
closed 2023-02-21 by ygoumaz