cyberbotics/webots_ros2 #299: Streaming viewer sets world into "pause"

anastasiapan, 2021-09-13:
**Describe the Bug**
I have created my own custom webots package with my own robot. I launch it with a launch file through a docker container. As soon as I connect with the streaming viewer the  world goes into "pause". I can see topics being published and the world is being displayed on the browser window but the world pauses. For example if I send a speed topic I can echo it inside the webots container but the robot doesn't move.

**Steps to Reproduce**
1. Run the image that launches the package:
`docker run -it --net host caprarobotics/webots:galactic`
2. Open the streaming viewer:
`file:///$WEBOTS_HOME/resources/web/streaming_viewer/index.html`
3. Click on `boradcast` and then connect
4. You will see the world in the browser window
5. But on the terminal you will see this:
```
[Aarhus.wbt" --batch --mode=realtime-1] Info: Sandboxing disabled by user.
[Aarhus.wbt" --batch --mode=realtime-1] pause

```

**Expected behavior**
The world should not go into pause mode and the interaction with the world should be displayed.

**System**
 - Webots Version: R2021b
 - ROS Version: Galactic 
 - Operating System: Windows 10, Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 2060
 
**Additional context**
From the galactic version I do not build these packages because they fail due to dependencies not available for ros2 galactic:
```
webots_ros2_driver 
webots_ros2_control 
webots_ros2_turtlebot
```
But the same thing happens with foxy version as well (pause once the streaming viewer connects).

lukicdarkoo, 2021-09-13:
> 3\. Click on `boradcast` and then connect

It seems there is a bug with the broadcast option. Can you avoid it?

@BenjaminDeleze It seems the bug is in the Webots core, maybe we should move the issue there

closed 2021-09-13 by BenjaminDeleze

BenjaminDeleze, 2021-09-13:
It should be fixed now.

Please let me now if it still not work.

(It is normal that for the world to be paused during the loading time of the simulation in the browser but it should resume once it is loaded)

anastasiapan, 2021-09-13:
@lukicdarkoo @BenjaminDeleze It works when disabling broadcast (pause while loading and then real-time) but it still remains on pause when broadcast is on. Do I need to  update anything?

BenjaminDeleze, 2021-09-13:
Normally not, it could be that your browser still have the faulty version in the cache, did you try to clear it ?

anastasiapan, 2021-09-15:
Thank you it worked!