cyberbotics/webots_ros2 #302: Can't compile from source and dependecy missing

NoeelMoeskops, 2021-09-28:
Hi,
trying to compile from source for galactic on Fedora 34 results in numerous compile errors. I think its mostly because webots_ros2 is not update with the upstream versions of their dependencies.
Some errors could be easy fixed like:
```
dev_ws/src/webots_ros2/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp: In memberfunctie ‘void webots_ros2_driver::Ros2Camera::publishRecognition()’:
dev_ws/src/webots_ros2/webots_ros2_driver/src/plugins/static/Ros2Camera.cpp:137:18: fout: ‘using ObjectHypothesisWithPose = struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’ {aka ‘struct vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >’} has no member named ‘id’
  137 |       hypothesis.id = std::string(objects[i].model);
      |                  ^~
```
by modifying line Ros2Camera.cpp:137 to:
```
hypothesis.hypothesis.class_id = std::string(objects[i].model);
```
The dependencies in `rosinstall_generator` in the Build & Install wiki is incomplete as it does not include "xacro control_toolbox". Also, ros2_control and ros2_controllers cannot be installed via rosinstall_generator and should be cloned manually in the src directory. 
Adding the missing dependencies manually still does not allow me to compile webots_ros2. Currently, I have either a whole bunch of invalid C statements(could be my fault, could not reproduce when writing this post). Or no matching function for `controller_manager::ControllerManager::update()`:
```
dev_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp: In memberfunctie ‘virtual void webots_ros2_control::Ros2Control::step()’:
dev_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:36:31: fout: no matching function for call to ‘controller_manager::ControllerManager::update()’
   36 |     mControllerManager->update();
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from dev_ws/src/webots_ros2/webots_ros2_control/include/webots_ros2_control/Ros2Control.hpp:29,
                 from dev_ws/src/webots_ros2/webots_ros2_control/src/Ros2Control.cpp:15:
dev_ws/src/install/controller_manager/include/controller_manager/controller_manager.hpp:132:37: note: candidate: ‘controller_interface::return_type controller_manager::ControllerManager::update(const rclcpp::Time&, const rclcpp::Duration&)’
  132 |   controller_interface::return_type update(
      |                                     ^~~~~~
```

Signature of `controller_manager::ControllerManager::update()` required parameters `Time` and `period`
```
controller_interface::return_type update( const rclcpp::Time & time, const rclcpp::Duration & period);
```
But I don't know what `update()` does since the [documentation](https://control.ros.org/api/classcontroller__manager_1_1ControllerManager.html) does not provide any information.
At this point I don't know what to do, revert to an old version of ros2_control that does not require parameters for the update function?
[This pull request](https://github.com/ros-controls/ros2_control/pull/520) together with [this commit](https://github.com/ros-controls/ros2_control/commit/ac3b10ea20104660f9ebf25b66ec04091773e086) seems to be the critical change.

Alright, so it seems to be updating periodically now instead of calling `update()` every time yourself, which is cool I guess. However, I don't know the code base enough to make changes. Reverting to [05db971](https://github.com/ros-controls/ros2_control/tree/05db9711f80be3de01ac7af3d4b136c304faf312) resulted in new errors 🙃:
```
--- stderr: joint_state_broadcaster                                                                                                                                                                                      
In file included from /home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:47:37: fout: ‘controller_interface::return_type joint_state_broadcaster::JointStateBroadcaster::update(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
   47 |   controller_interface::return_type update(
      |                                     ^~~~~~
/home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:51:18: fout: ‘rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn joint_state_broadcaster::JointStateBroadcaster::on_init()’ marked ‘override’, but does not override
   51 |   CallbackReturn on_init() override;
      |                  ^~~~~~~
In file included from /home/noeel/ros2_galactic/install/class_loader/include/class_loader/class_loader_core.hpp:57,
                 from /home/noeel/ros2_galactic/install/class_loader/include/class_loader/class_loader.hpp:55,
                 from /home/noeel/ros2_galactic/install/pluginlib/include/pluginlib/class_list_macros.hpp:40,
                 from /home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:281:
/home/noeel/ros2_galactic/install/class_loader/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = joint_state_broadcaster::JointStateBroadcaster; B = controller_interface::ControllerInterface]’:
/home/noeel/ros2_galactic/install/class_loader/include/class_loader/meta_object.hpp:216:7:   required from here
/home/noeel/ros2_galactic/install/class_loader/include/class_loader/meta_object.hpp:218:12: fout: invalid new-expression of abstract class type ‘joint_state_broadcaster::JointStateBroadcaster’
  218 |     return new C;
      |            ^~~~~
In file included from /home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:34:7: note:   because the following virtual functions are pure within ‘joint_state_broadcaster::JointStateBroadcaster’:
   34 | class JointStateBroadcaster : public controller_interface::ControllerInterface
      |       ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:24,
                 from /home/noeel/dev_ws/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/noeel/dev_ws/install/controller_interface/include/controller_interface/controller_interface.hpp:98:23: note:     ‘virtual controller_interface::return_type controller_interface::ControllerInterface::update()’
   98 |   virtual return_type update() = 0;
      |                       ^~~~~~
```
I did not investigate further, and I'm hoping you folks can help me out.

Also, as a side note: Colcon (package webots_ros2_driver) was also complaining that it needed exactly python 3.8. I have 3.8 together with 3.9.x installed on my system. However, Cmake found 3.9 before 3.8 and aborted the build because of a version mismatch.
```
--- stderr: webots_ros2_driver                                                                                                                                                                                     
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonLibs: Found unsuitable version "3.9.6", but required
  is exact version "3.8" (found /usr/lib64/libpython3.9.so)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:592 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindPythonLibs.cmake:310 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:21 (find_package)
```
But this seems more like a Cmake issue though. Not sure what you can do about it.

lukicdarkoo, 2021-09-29:
ROS 2 is not fully supported on Fedora and ROS Galactic is not supported by ros2_control (therefore the new version of webots_ros2 is not supported either).

Unfortunately, ros2_control is not released for Galactic, although it seems it will be released soon:
https://github.com/cyberbotics/webots_ros2/pull/250

You can ignore that package:
```
colcon build --packages-skip webots_ros2_control
```
(it will cause many examples to fail as most of them depend on webots_ros2_control)

What about going with Foxy? It will have longer support and it has more packages.

> Also, as a side note: Colcon (package webots_ros2_driver) was also complaining that it needed exactly python 3.8. I have 3.8 together with 3.9.x installed on my system. However, Cmake found 3.9 before 3.8 and aborted the build because of a version mismatch.

Do you have Python 3.8 header files? On Ubuntu, you can install the files as:
```
sudo apt install libpython3.8-dev
```

NoeelMoeskops, 2021-09-29:
> Do you have Python 3.8 header files?

Yes, /usr/lib64/libpython3.8.so (sys link) and /usr/lib64/libpython3.8.so.1.0

I thought Webots was compatible with galactic? Ooh well, skipping webots_ros2_control resulted in errors in the diff_drive_controller from ros2_controllers package. But that's out of your control I guess.
```In file included from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp:18:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:37:41: fout: expected ‘)’ before ‘rolling_window_size’
   37 |   explicit RollingMeanAccumulator(size_t rolling_window_size)
      |                                  ~      ^~~~~~~~~~~~~~~~~~~~
      |                                         )
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:61:3: fout: ‘size_t’ does not name a type
   61 |   size_t next_insert_;
      |   ^~~~~~
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:24:1: note: ‘size_t’ is defined in header ‘<cstddef>’; did you forget to ‘#include <cstddef>’?
   23 | #include <vector>
  +++ |+#include <cstddef>
   24 | 
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp: In memberfunctie ‘void diff_drive_controller::RollingMeanAccumulator<T>::accumulate(T)’:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:44:21: fout: ‘next_insert_’ was not declared in this scope
   44 |     sum_ -= buffer_[next_insert_];
      |                     ^~~~~~~~~~~~
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp: In memberfunctie ‘T diff_drive_controller::RollingMeanAccumulator<T>::getRollingMean() const’:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:54:5: fout: ‘size_t’ was not declared in this scope; did you mean ‘std::size_t’?
   54 |     size_t valid_data_count = buffer_filled_ * buffer_.size() + !buffer_filled_ * next_insert_;
      |     ^~~~~~
      |     std::size_t
In file included from /usr/include/c++/11/cassert:43,
                 from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:22,
                 from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp:18:
/usr/include/c++/11/x86_64-redhat-linux/bits/c++config.h:2426:33: note: ‘std::size_t’ declared here
 2426 |   typedef __SIZE_TYPE__         size_t;
      |                                 ^~~~~~
In file included from /usr/include/c++/11/cassert:44,
                 from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:22,
                 from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp:18:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:55:12: fout: ‘valid_data_count’ was not declared in this scope
   55 |     assert(valid_data_count > 0);
      |            ^~~~~~~~~~~~~~~~
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp: In memberfunctie ‘virtual void TestAccumulator_test_accumulator_Test::TestBody()’:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp:27:64: fout: no matching function for call to ‘diff_drive_controller::RollingMeanAccumulator<double>::RollingMeanAccumulator(int)’
   27 |   diff_drive_controller::RollingMeanAccumulator<double> accum(4);
      |                                                                ^
In file included from /home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/test/test_accumulator.cpp:18:
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:34:7: note: candidate: ‘diff_drive_controller::RollingMeanAccumulator<double>::RollingMeanAccumulator()’
   34 | class RollingMeanAccumulator
      |       ^~~~~~~~~~~~~~~~~~~~~~
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:34:7: note:   candidate expects 0 arguments, 1 provided
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:34:7: note: candidate: ‘diff_drive_controller::RollingMeanAccumulator<double>::RollingMeanAccumulator(const diff_drive_controller::RollingMeanAccumulator<double>&)’
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:34:7: note:   no known conversion for argument 1 from ‘int’ to ‘const diff_drive_controller::RollingMeanAccumulator<double>&’
/home/noeel/dev_ws/src/ros2_controllers/diff_drive_controller/include/diff_drive_controller/rolling_mean_accumulator.hpp:34:7: note: candidate: ‘diff_drive_controller::RollingMeanAccumulator<double>::RollingMeanAccumulator(diff_drive_controller::RollingMeanAccumulator<double>&&)’
[... cut ...]
gmake[2]: *** [CMakeFiles/test_accumulator.dir/build.make:76: CMakeFiles/test_accumulator.dir/test/test_accumulator.cpp.o] Fout 1
gmake[1]: *** [CMakeFiles/Makefile2:129: CMakeFiles/test_accumulator.dir/all] Fout 2
```
Skipping more packages in an attempt to get something to work results on dependency breaking. Up to the point that`colcon build --packages-skip webots_ros2_control diff_drive_controller webots_ros2_epuck webots_ros2_turtlebot webots_ros2_universal_robot webots_ros2_tests webots_ros2` was needed to avoid any errors.

But at this point I'm wondering if this is even functional...

BenjaminHug8, 2021-10-01:
> Ooh well, skipping webots_ros2_control resulted in errors in the diff_drive_controller from ros2_controllers package. But that's out of your control I guess.

Yes, by skipping `webots_ros2_control` it's not surprising to get those errors. 

> Skipping more packages in an attempt to get something to work results on dependency breaking. Up to the point that`colcon build --packages-skip webots_ros2_control diff_drive_controller [...]

Exactly, you won't be able to make it work as you need `webots_ros2`.

lukicdarkoo, 2021-10-01:
Some progress on this:
https://github.com/ros-controls/ros2_control/issues/314#issuecomment-931440974

BenjaminHug8, 2021-10-13:
> The instalation documentation for installing the dependecies for Foxy is also incomplete. The full `rosinstall_generator` arguments to install all dependecies is as follows:

I am a bit confused as `rosinstall_generator` is for Windows dependency, and if you are on Linux you should just have to run:

```
source /opt/ros/$ROS_DISTRO/local_setup.bash

# Retrieve the sources
cd /path/to/ros2_ws
git clone --recurse-submodules -b $ROS_DISTRO https://github.com/cyberbotics/webots_ros2.git src/webots_ros2

# Check dependencies
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO

# Building packages
colcon build

# Source this workspace (careful when also sourcing others)
source install/local_setup.bash
```
Does that not work for you ?

closed 2022-03-07 by BenjaminHug8