cyberbotics/webots_ros2 #303: Windows installation dependencies - rosinstall_generator do not work for ros2_control and ros2_controllers

BenjaminHug8, 2021-10-01:
**Describe the Bug**
Past in the documentation the following commands were supposed to work to install `ros2_control` and `ros2_controllers`.

```
rosinstall_generator ros2_control ros2_controllers --rosdistro foxy > deps.repos
vcs import src < deps.repos
```
Instead we have to clone them from the `foxy` branch.

**To do**

- [ ] Check if these packages can be installed with rosinstall_generator.

lukicdarkoo, 2021-11-19:
I tried to make it work a while ago, but it seems there is a ROS bug:
https://github.com/ros2/ros_workspace/pull/21#issuecomment-888062636

Let me give it a second try

lukicdarkoo, 2021-11-23:
1. Creates an empty `deps.txt` file:
    ```
    rosinstall_generator webots_ros2 --exclude RPP --deps > deps.txt
    ```
2. Includes dependencies even though some of them already exist in `src`:
    ```
    set ROS_PACKAGE_PATH=C:\dev\ros2_foxy
    rosinstall_generator webots_ros2 --exclude RPP --deps > deps.txt
    ```
3. Creates an empty `deps.txt` file:
    ```
    set ROS_PACKAGE_PATH=C:\Users\lukic\ros2_ws
    rosinstall_generator webots_ros2 --exclude RPP --deps > deps.txt
    ```

Looking for a way to exclude ROS 2 core packages and already cloned packages from the `deps.txt` file...

lukicdarkoo, 2021-11-23:
@BenjaminHug8 Solved, checked the installation procedure now

BenjaminHug8, 2021-11-23:
Perfect thank you.

closed 2021-11-23 by BenjaminHug8